arendst/Tasmota #81: no user-interface.h

consolacion, 2017-02-18:
file 'support.h'  asks for 'user_interface.h'
but there is no such file  should it be ' user_config.h' perhaps?

hemantkamalakar, 2017-02-24:
same issue.

sketch\support.h:8:28: fatal error: user_interface.h: No such file or directory

 #include "user_interface.h"

lucwuyts, 2017-02-24:
This file is included in the SDK 2.3.0
Maybe install the Arduino ESP8266 project again?

hemantkamalakar, 2017-02-24:
Yes. It helped.
Thanks a lot for swift reply.

closed 2017-03-08 by arendst

joann235, 2018-05-24:
How can this be closed if the error is still occurring?  I flashed a sonoff basic with no problem.  Put another one on immediately after to flash and got this error.  Nothing changed in between.  There must be a more fundamental problem and I have no idea what sdk 2.30 is.