arendst/Tasmota #82: Please add support for wion plug with power monitoring

davidelang, 2017-02-19:
This is an esp8266 based outlet available for $25 that does power monitoring

http://www.homedepot.com/p/Woods-WiOn-Indoor-Wi-Fi-Current-Tap-with-2-USB-Charging-Ports-Wireless-Switch-and-Programmable-Timer-50055/206758644
https://www.amazon.com/WiOn-50055-Indoor-Monitor-Wireless/dp/B00ZYLUBJU/ref=sr_1_4?s=hi&ie=UTF8&qid=1487485998&sr=1-4&keywords=wion

an alternate firmware has already been created for the simple switching:

http://thegreatgeekery.blogspot.com/2016/02/ecoplug-wifi-switch-hacking.html
with source available on github under GPLv3
https://github.com/scottjgibson/esp8266Switch

LED = 2
relay = 15
button = 13

in the comments at the greatgeedery blog, there is code to do the power monitoring (clock on GPIO0, data on GPIO12)

If you would like me to purchase samples of this line for you to use, contact me with your address and I'll purchase some on Amazon and have them shipped to you

arendst, 2017-02-19:
David, as I live in Europe our plugs are different.

But you can try it yourself by tweaking some values at the bottom of file ``sonoff_template.h``. From Scott Gibsons GitHub code (switch only, no power monitoring, different hardware) I conclude the following parameters should make yours work:
```
  { "User Test",       // Sonoff Basic User Test
     GPIO_USER,        // GPIO00 Optional sensor (pm clock)
     0,
     GPIO_LED1,        // GPIO02 Green Led (1 = On, 0 = Off)
     0, 0, 0, 0, 0, 0, 0, 0, 0,
     GPIO_USER,        // GPIO12 Optional sensor (pm data)
     GPIO_KEY1,        // GPIO13 Button
     0,
     GPIO_REL1,        // GPIO15 Relay (0 = Off, 1 = On)
     0
  }
```
Just replace the current settings for ``User Test`` with the ones above and give it a go.

davidelang, 2017-03-16:
Thanks for merging the basic support. I have a branch that is fetching the sensor data and outputs it as 4 hex values and 4 decimal values (4 bytes each), I haven't made much progress decoding them (any assistance would be appriciated) I posted a bunch of the raw data at https://bitbucket.org/xoseperez/espurna/issues/48/ecoplug-power-sensing where the question was also raised

davidelang, 2017-03-18:
I picked up a varister and did some power monitoring on this device at different voltages, the results are posted at (http://lang.hm/wion) if anyone is able to make any progress decoding the data

the setup is with the wion plugged into the sonoff pow and both logging every few seconds. the values on the wion get reset every time they are read.

It seems that these devices get very unhappy at ~40v AC

closed 2018-07-10 by ascillato2