arendst/Tasmota #83: Any chance of extending the IRremote code with Mitsubishi aircon support?

mihalski, 2017-02-19:
Hi,

Any chance of adding support for Mitsubishi Air conditioning remotes? It would be fantastic in conjunction with the temp/humidity monitoring. There is code available at https://github.com/r45635/HVAC-IR-Control.

Regards,
Michal

mihalski, 2017-02-23:
Nobody here has a Mitsubishi Electric air conditioning unit? I've managed to test the library I linked to on an ESP8266 and it does work.

Here's a more specific link:

https://github.com/r45635/HVAC-IR-Control/blob/master/HVAC_ESP8266/HVAC_ESP8266.ino

Maybe if that could be added as a protocol to choose from? "MitsubishiHVAC" ? And then you can send the settings rather than the codes via console/MQTT. i.e. something along the lines of:

IRSend { "protocol": "MitsubishiHVAC", "Mode": COLD, "Temp": 23, "Fan": Silent, "Vanne": Auto, "WideVanne": Middle, "Area": Swing,  "Power": On }

Regards,
Michal

alesoblak, 2017-02-25:
I have Mitcubishi air coniditioning unit. Will check this thread regularly

mihalski, 2017-02-25:
I've prompted a contributor of the IR lib to have a look into adding the Mitsubishi support there so maybe we'll get it via that if not directly in this firmware.

It would probably a better solution ultimately as it would let both projects focus on their own projects.

altelch, 2017-02-25:
That would be a better Solution. Problem might be that Arduino includes a forked version of this lib. But after Theo including the libs in a separate folder thats easy to overcome.

mihalski, 2017-02-25:
FYI.

https://github.com/markszabo/IRremoteESP8266/issues/70

davidelang, 2017-03-27:
I've seen e-mails on this topic, has it been fixed yet?

mihalski, 2017-03-27:
See https://github.com/arendst/Sonoff-Tasmota/pull/257

It is fully working (short of supporting a temp of 16C on the remote), but that's not a significant issue.

davidelang, 2017-03-27:
That's what I thought, but I wanted to confirm before closing it.

mihalski, 2017-03-27:
Mind you. It still hasn't been added in the official firmware :/

davidelang, 2017-03-27:
is there value in keeping this issue open until it is added?

Is the delay in getting it upstream or getting the newer upstream version into tasmota?

mihalski, 2017-03-27:
I honestly don't know what the reason for the delay is.

davidelang, 2017-03-27:
is the code in the IR library now?

mihalski, 2017-03-27:
Yes. It has been since the 0.2 release 26 days ago.

They're now on 1.0.2 and working well with the Arduino library package manager updating.

arendst, 2017-04-02:
will be implemented in next release

closed 2017-04-29 by arendst