arendst/Tasmota #84: can Gpio switch not direct action?

hamwong, 2017-02-19:
can Gpio switch not direct perform relay action and publish status to MQTT?

davidelang, 2017-02-19:
I think it does that if it's not able to reach a mqtt server

hamwong, 2017-02-19:
on my test environment, I got MQTT online, only result is switch direct turn on/off relay and post MQTT,
I wish there is an option don't switch relay and post MQTT, let server - openhab decide on or off,

davidelang, 2017-02-19:
you can set the button to send a different message, that's what the switchtopic 
is for.

set the button to send a different topic and openhab will make the decision.

hamwong, 2017-02-20:
Thanks @davidelang, as too many commands, and not much description on command usage, I didn't aware switchtopic for this usage, another issue come, is this is a bug or limit as if I got 2 gpio set for same function(switch). only one of gpio will work, other one won't function,

I set as below:
{"GPIO3":9 (Switch), "GPIO4":0 (None), "GPIO14":3 (Switch)}
than only GPIO14 works, GPIO3 no response, 

I like to set it make it works as below condition, GPIO14 connect to HC-SR501 PIR and work with MQTT (openhab),  and GPIO3 work with wall switch, so that user may manually overide.

as far as I test, I only may work out as switch + DHT22 on GPIO 3,14, but not both switch.

Nayar, 2017-02-20:
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/158

I injected some codes into the firmware that read the GPIO then published on `/stats/sonoff/GPIO` topic. Then I could implement this: https://nayarweb.com/blog/2017/iot-series-smart-lightsensors-in-the-kitchen/

I need to rewrite the patch and maybe merge it upstream

hamwong, 2017-02-21:
Hi @Nayar
Thanks, I am so excite that this feature will come, I look at #158, does that mean I can assign GPIO14 connect to HC-SR501 PIR and work with MQTT (openhab), and GPIO3 work with wall switch (direct control)?

closed 2017-02-28 by hamwong