arendst/Tasmota #85: WS2812 independent switching

IngoOWL, 2017-02-19:
Bug or feature?
When switching the relays (basic, 2chan. 4chan), a connected WS2812 is also switched on and off. Is it possible to do this independently?
With the user-defined settings, a switch is listed under 9, can this be switched independently via an http get?

altelch, 2017-02-19:
Currently it's a feature. Theo implemented it this way maybe with Sonoff LED in mind. I'd like the idea of virtual devices with own power state so you could even segment your LED strip and switch the segments on their own. The switch on 9 is for defining the gpio as wall switch input.

stoehrmark, 2017-03-16:
**+1 to  independent switching.**
 I have an Electrodragen i want to control a Costco Led light and add some WS2812 for some fun color, I need the relay and WS2812  independent. 

I have been using some excelent work by kitesurfer1404 at GitHub. The WS2812FX code works very well. Maybe it could some how be added to the Sonoff-Tasmota code,

[https://github.com/kitesurfer1404/WS2812FX](url)

Costco Lamp [https://www.costco.ca/Cloudraker-LED-Ceiling-Fixture.product.100278984.html](url)
[image]

davidelang, 2017-03-16:
I don't understand, isn't the control of any WS* LEDs independent of any relays 
already?

khcnz, 2017-03-16:
@davidelang yes agreed. 

@stoehrmark if you use the WS* then you would communicate with that chip (via I2C?) not using any of the "LED" pins in tasmota. 

@altelch I have been working on a branch that lets you set an individual GPIO_USER to be a PWM output where you can control the level through a command. This would therefor allow independent switching of any output on/off uncoupled from the switch/relay state. PWM allows 1023 unique voltage levels but if you set it to the maximum analog output I presume this is the same as setting the digital out to 1....

stoehrmark, 2017-03-18:
Thanks for the response. Maybe i am doing something wrong. 

Here is my test setup and i made a video of what happens when i click the buttons.
 The Ws2812s are attached to GPIO3 Serial In. I also get power from the Electrodragon but will use external 5V in the final setup. When clicking relay 1 the WS2812s go on/off. With clicking relay 2 they flash.

https://youtu.be/4ts_Je5-fys

davidelang, 2017-03-27:
can you hook the LEDs to a different GPIO, I can't think of why clicking the relay would change serial in, but if you were actually connected to serial out and have serial logging happening, that would explain things

davidelang, 2017-04-23:
closing due to the lack of response. If you are still working on this issue, please re-open it.

closed 2017-04-23 by davidelang

joecotton, 2017-04-25:
I was able to see why the relay and WS2812 strip control are not independent. In xdrv_ws2812.ino, several of the functions check or set the power variable, which then propagates to the relay. Perhaps those parts could be masked in a 

if (sysCfg.module == SONOFF_LED) {}

statement so they only take effect if the module type is the Sonoff LED.