arendst/Tasmota #86: 3.9.19 will not compile IRremoteESP8266 issue

Plum67, 2017-02-19:
Tried to Compile 3.8.18 & 3.9.19 and always get error message below. What do I need to do to get it to compile I have added all the libraries but it looks like something in IRremoteESP8266, btu I am new to all this and I do not know what to do to get it to compile?

Arduino: 1.8.0 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\libraries\IRremoteESP8266-master\IRremoteESP8266.cpp: In member function 'void IRsend::sendKelvinator(unsigned char*)':

C:\Program Files (x86)\Arduino\libraries\IRremoteESP8266-master\IRremoteESP8266.cpp:625:14: error: 'KELVINATOR_STATE_LENGTH' was not declared in this scope

   for (; i < KELVINATOR_STATE_LENGTH; i++)

              ^

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Cartman16, 2017-02-19:
Same Issue here!

hoekbrwr, 2017-02-19:
Find the library [IRRemoteESP8266.h](https://github.com/sebastienwarin/IRremoteESP8266) and install it in Arduino IDE Sketch/Library Manager

davidelang, 2017-02-19:
I fixed this by modifying the file IRremoteESP8266.cpp and adding the line

'#include "IRKelvinator.h"'

I don't have an IR transmitter available to use, but this let it compile and run on a couple units.

hoekbrwr, 2017-02-19:
It seems that there are things added to this IRRemote further on adter my version 3.9.17. Only Theo knows what has been added that causes this .

altelch, 2017-02-20:
Looks like you directly installed IRRemoteESP8266 from github and not the Version included into the Arduino library manager. The Kelvinator was added about 22 days ago and doesn't exist in the Arduino managed version. So it has nothing todo with Theo but with the source of your library and it's version.

Plum67, 2017-02-20:
So what version are we supposed to use?

hoekbrwr, 2017-02-20:
Yes I did took the original from the link, because it wa not yet in the library. I tried again to compile 3.9.19, no problem!

altelch, 2017-02-20:
Arduino IDE -> Sketch -> Library... -> Manage Libraries, search for IRremoteESP8266 and you will find a 1.0.0 version from https://github.com/sebastienwarin/IRremoteESP8266 (which is a fork). This is the "official" Arduino Version.

hoekbrwr, 2017-02-20:
Strange it does not find the latest version!, Which method to use should not be any difference?

altelch, 2017-02-20:
The arduino IDE can't find the latest version as it uses a fork of sebatienwarin labeled 1.0.0 and not the original marc szabo version which seems to have no versioning (what might be the problem for the arduino guys).

hoekbrwr, 2017-02-20:
As @Plum67 already noticed. What version we are supposed to use? This is not only a problem with the IR lib, but is general. I think there must be a notice from the author who included the library in the first place. However one cannot be aware of forks of unofficial versions. I am absolutely not a software engineer and have never used Github for SW versioning, but I can handle the outcomes of examples and find me the accompanying libs on the net. But I have no idea about other versions. However the author of the "main" SW should point to the lib the SW has been tested with. All confusing???!!!???

altelch, 2017-02-20:
All additional libraries are the Versions included in Arduino Library Manager (always tested with the latest version). I never had to install a library from github or somewhere else using a zip file. 

https://www.arduino.cc/en/Guide/Libraries

In the Library Manager use the Filter to find the missing libs and install the latest version (default).

arendst, 2017-02-20:
Sorry for the trouble you seem to be in.

One of the reasons I implemented my own drivers for DHT and DS18B20 was to not get involved with library issues. Library incompatibility issues are easy to occur as they are not managed centrally and everyone can fiddle around with them. That's also the reason why I stayed at PubSubClient from Nick O'Leary, a stable library with some issues to overcome locally.

In the last months extra features were suggested and implemented using libraries that now seem to forget their legacy users resulting in your experience.

Another problem is the 'force' of people wanting to use platformIO also used by Travis, the github online recompile service I use instead of Arduino IDE bringing it's own problems regarding library versions.

So for now Travis/platformIO is using the IRremoteESP8266 from https://github.com/sebastienwarin/IRremoteESP8266 version fee16e880b which seems to be rather stable as it hasn't changed for over 12 months. I downloaded and installed the zip file in my standalone Arduino IDE environment (portable/sketchbook/libraries/IRremoteESP8266) and use it.

So your options now are:
- Disable IRremote by option USE_IR_REMOTE in user_config.h.
- Install the correct library from the github link above.

I'll update the wiki with version information regarding the used libraries.

hoekbrwr, 2017-02-20:
I am not planning to use Platformio and stick to Arduino IDE and the way @altelch suggested way of including libraries. At the moment it is not an issue for an IRRemote but only the principle. I leave it now and go on doing other things. Thanks for your contributions to this subject.

0n3-70uch, 2017-02-20:
Hello,

I checked the new wiki writings but I have one question:

> Download and unzip the pubsubclient MQTT library version 2.6 into directory 

For me there is no version 2.6 only 2.4. Is it possible that you mean version 2.4?

Greetings
Fabi

arendst, 2017-02-20:
Looking in file https://github.com/knolleary/pubsubclient/blob/master/library.properties it says 2.6 as does the Arduino library manager once installed.

0n3-70uch, 2017-02-20:
Oh right. I looked only in this file:
https://github.com/knolleary/pubsubclient/blob/master/CHANGES.txt

And sry for the next question. Where can I find the version 2.2.6 of NeoPixelBus. Because I only see this sort of version name: "85b250d"

arendst, 2017-02-20:
Again, look in the https://github.com/Makuna/NeoPixelBus/blob/master/library.properties file for the version.

Once a monkey knows a trick....

0n3-70uch, 2017-02-20:
Haha yes thank you... :-) I will never ask the same question again ;)

arendst, 2017-02-20:
Just updated links in Wiki to library releases if available.

davidelang, 2017-02-20:
On Mon, 20 Feb 2017, Theo Arends wrote:

> In the last months extra features were suggested and implemented using 
> libraries that now seem to forget their legacy users resulting in your 
> experience.

This sort of lack of backwards compatibility is very common in FOSS projects, 
few people do the sort of thing you have done of creating a new project when 
they go to make backwards incompatible changes.

I have three libraries installed via git because the instructions on the wiki 
had links to the github repos and just said that they needed to be installed. So 
if those are not the versions that you want us to use, please update the wiki to 
point us at the right versions.

since it seems that I have things running now, I'm going to stick with the 
github versions unless they cause me too much grief.

In this case, a missing include was easy to fix.

yann25, 2017-02-20:
Hey, 

I sorry for my noob question, but what do you mean with "Disable IRremote by option USE_IR_REMOTE in user_config.h." ?

Do you mean comment it out? Because that did not work for me...

arendst, 2017-02-22:
If you copy the lib folder into your platformIO project folder you should be fine.

closed 2017-02-22 by arendst