arendst/Tasmota #88: multiple switchtopics?

davidelang, 2017-02-19:
I find that I have a need to create some additional inputs (that will then be used to control different devices), and I see that switchtopic can be set to change the main button to send some other mqtt message. 

If I add additional switches to the available GPIOs, is there a way to set switchtopics for them?

arendst, 2017-02-20:
I did add a number after the defined switchtopic so that should make identification possible.

davidelang, 2017-02-20:
as in switchtopic<x> (similar to power<x>)?

arendst, 2017-02-20:
I thought soo but after looking at the code it seems to do:
- Switch1: cmnd/<switchtopic>/POWER1
- Switch2: cmnd/<switchtopic>/POWER2
etc.
and all depending on the amount of devices (relays) are present so it might not work the way you want it to...

BTW how did you select more than one switch as in the default code I do not allow more than one switches?

hamwong, 2017-02-20:
@arendst does the firmware allow multi switch like below?
{"GPIO3":9 (Switch), "GPIO4":0 (None), "GPIO14":9 (Switch)}
as I only can get first GPIO work with swtich, second assign to swith with no response on hardware

davidelang, 2017-02-20:
On Mon, 20 Feb 2017, Theo Arends wrote:

> BTW how did you select more than one switch as in the default code I do not allow more than one switches?

I haven't yet, I'm wanting to do so. For example, the sonoff sv has a 9 pin 
header on it that exposes 3 GPIOs, I want to hook switches to all three to 
control other things in the house.

mihalski, 2017-02-24:
I've just hit this limitation by wanting to connect 2 PIR sensors (facing in opposite directions). Is this something that can be achieved?

i.e. the Idea being that these PIR's switch on lights for different areas via MQTT.

hamwong, 2017-02-24:
do you mind share how do you do it?
I think firmware limit only each type only can one be operate, 
do you mind share how do you break the same type single device limit?

arendst, 2017-02-24:
Did some rewrite for the switches. It now supports four of them but beware of the functionality: without switchtopic set it tries to control the relays depending on the amount of relays present. Best usage is setting a switchtopic and use MQTT to react. SwitchMode1 to SwitchMode4 is also available.

Implemented in version 3.9.21 just released

hamwong, 2017-02-24:
it works, Thanks for your great works, just one question,
 how to define switch1 as sonoff direct response and switch 2 as mqtt?
as I can see now all on mqtt, can difference switch have difference output?

hamwong, 2017-03-09:
U first need to set 2 switch as difference switch
Like gpio3 switch1 gpio14 switch2
Then set switchtopic1 123
Switchtopic2 234
Than 2 switch have difference mqtt topic

hamwong, 2017-03-09:
You also can define each switch reaction by switchmode1 by check status
Switchmode1 1 by Change assume its 0 or 2 by default

davidelang, 2017-03-09:
you only have one power because you would only have one relay (and I'm not sure 
tasmota supports zero relays)

what you would need to do is

switchmode1 mode
switchmode2 mode

where mode is 0-4

search for switchmode on the command page 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

David Lang

davidelang, 2017-03-09:
oops, I answered switchmode not switchtopic, but the basis is the same, as hamwong correctly answered.

mihalski, 2017-03-09:
Thank you! it FINALLY works!

I SWEAR TO GOD I had done just this in the past.. Perhaps I had messed up something the last 20 times I tried it but I can't imagine what :/

hamwong, 2017-03-09:
Doesn't matter, human mistake happens all the time as computer never go wrong, if it did, that's human mistake XD

mihalski, 2017-03-09:
Wait.. something is wrong..
```
19:23:04 MQTT: cmnd/stairs/POWER1 = ON
19:23:16 MQTT: cmnd/stairs/POWER2 = OFF
19:23:19 MQTT: cmnd/stairs/POWER2 = ON
19:23:47 MQTT: cmnd/stairs/POWER1 = OFF
19:24:14 MQTT: cmnd/stairs/POWER2 = OFF
19:24:17 MQTT: cmnd/stairs/POWER2 = ON
```
I'm getting two switches on the one topic..

Even though I set it up with:
```
19:09:46 CMND: switchtopic2 tvarea
19:09:46 MQTT: stat/upstairs/RESULT = {"SwitchTopic":"tvarea"}
19:10:00 CMND: switchtopic1 stairs
19:10:00 MQTT: stat/upstairs/RESULT = {"SwitchTopic":"stairs"}
```
The tvarea topic is never triggered.

davidelang, 2017-03-09:
> Wait.. something is wrong..
> ```
> 19:23:04 MQTT: cmnd/stairs/POWER1 = ON
> 19:23:16 MQTT: cmnd/stairs/POWER2 = OFF
> 19:23:19 MQTT: cmnd/stairs/POWER2 = ON
> 19:23:47 MQTT: cmnd/stairs/POWER1 = OFF
> 19:24:14 [...]

Ok, that's not what I expected to have happen, but that is separating the two 
switches so that you can handle them separately on the server.

@arendst, is this working as expected and we have been misunderstanding 
swtichtopic<n>? or is this a bug?

David Lang

arendst, 2017-03-09:
Fact: There is but ONE switchtopic available that can be addressed by ``switchtopic``, ``switchtopic1``, ``switchtopic2`` or even ``switchtopic1984``. Setting switchtopic to ``upstairs`` provides responses like
```
cmnd/upstairs/POWER1 for switch1
cmnd/upstairs/POWER2 for switch2
etc.
```
So the difference in switches is the number after POWER\<x\> as I tried to explain in the fourth entry of this issue (where it failed to note the switchtopic due to my mistake of using \< and \>).

I must say you all came a long way after 17 days ;-)

mihalski, 2017-03-09:
Well I got POWER1 and POWER2 to appear.

It wasn't appearing before I used the switchtopic commands as seen below:
```
18:55:56 MQTT: stat/upstairs/RESULT = {"POWER":"ON"}
18:55:56 MQTT: stat/upstairs/POWER = ON
18:56:51 MQTT: stat/upstairs/RESULT = {"POWER":"OFF"}
18:56:51 MQTT: stat/upstairs/POWER = OFF
18:56:55 MQTT: stat/upstairs/RESULT = {"POWER":"ON"}
18:56:55 MQTT: stat/upstairs/POWER = ON
18:57:13 CMND: switchtopic1 123
18:57:13 MQTT: stat/upstairs/RESULT = {"SwitchTopic":"123"}
18:57:23 MQTT: tele/upstairs/STATE = {"Time":"2017-03-09T18:57:23", "Uptime":38, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":58}}
18:57:31 CMND: Switchtopic2 234
18:57:31 MQTT: stat/upstairs/RESULT = {"SwitchTopic":"234"}
18:57:50 MQTT: cmnd/234/POWER2 = OFF
18:57:53 MQTT: cmnd/234/POWER2 = ON
18:58:09 MQTT: cmnd/234/POWER1 = ON
18:58:49 MQTT: cmnd/234/POWER2 = OFF
18:58:52 MQTT: cmnd/234/POWER2 = ON
18:58:52 MQTT: cmnd/234/POWER1 = OFF
18:59:48 MQTT: cmnd/234/POWER2 = OFF
18:59:51 MQTT: cmnd/234/POWER2 = ON
19:00:46 MQTT: cmnd/234/POWER2 = OFF
19:00:49 MQTT: cmnd/234/POWER2 = ON
```
That was me following the original example.

Is that a bug or intended behaviour?

mihalski, 2017-03-09:
OK. Granted.. This is 3.9.22 and I'm about to upgrade it to the latest (only stayed on this version because this module was sitting in limbo until I figured out how to configure it). However, in case this is of any interest, this just happened:
```
20:53:11 MQTT: cmnd/stairs/POWER2 = OFF
20:53:15 MQTT: cmnd/stairs/POWER2 = ON
20:53:34 CMND: switchtopic upstairs
20:53:34 MQTT: stat/upstairs/RESULT = {"SwitchTopic":"upstairs"}
20:54:10 MQTT: cmnd/upstairs/POWER2 = OFF
20:54:10 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:54:10 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:54:10 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
20:54:10 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:54:16 MQTT: cmnd/upstairs/POWER2 = ON
20:54:16 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:54:16 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:54:16 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
20:54:16 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:54:40 MQTT: cmnd/upstairs/POWER1 = ON
20:54:40 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:54:40 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:54:40 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
20:54:40 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:55:11 MQTT: cmnd/upstairs/POWER2 = OFF
20:55:11 MQTT: stat/upstairs/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:55:11 MQTT: stat/upstairs/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:55:11 MQTT: stat/upstairs/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, WebPassword, Emulation, SwitchMode"}
20:55:11 MQTT: stat/upstairs/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
```
With every OFF or ON. So if whatever this is has been fixed, great. If not, maybe this will be helpful.

arendst, 2017-03-09:
As you are running WITHOUT a relay it is not a sonoff and therefore anything could happen but apparently I seem to have provided some nice features anyhow. It's intended behavior:
- A switch was intended to control the corresponding physical relay in a sonoff OR send a MQTT message
- Without relays the software assumes no real devices are present (Maxdevice = 0) but a single virtual relay is present and configures itself for it.
- Using any switch in that situation WITHOUT naming a switchtopic will control the single virtual relay and responds with what you see as ``stat/upstairs/POWER ON``. That's POWER without a number as it assumes there is only one relay present.
- Setting switchtopic to anything but 0 starts using another part of the code and allows for sending MQTT messages like ``cmnd/switchtopic/POWER1 ON`` where the number after POWER is the switch number.

davidelang, 2017-03-09:
@arendst 
If this was on a module with a relay, would it still give just POWER without swtichtopic being set? or with Maxdevice=1 would it do POWER1/POWER2 without needing to set the switchtopic?

davidelang, 2017-03-09:
I added the page https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs to try and explain how to wire additional things (like this) to a sonoff and https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types to provide hints on how to define a new module type.

arendst, 2017-03-09:
@davidelang If there are more relays it will start using numbers like POWER1 and POWER2 etc. BUT it will also control these relays.

davidelang, 2017-03-09:
On Thu, 9 Mar 2017, Theo Arends wrote:

> @davidelang If there are more relays it will start using numbers like POWER1 and POWER2 etc. BUT it will also control these relays.

If there is only one relay, and you hook a second switch, will it do 
POWER1/POWER2, or is it only when you have two or more relays that it will add a 
number to the POWER?

arendst, 2017-03-09:
Only when you have two or more relays. With one relay and two switches the second switch will control the single relay too.

davidelang, 2017-03-09:
something that just occured to me, the user switch1-switch4, are those the 
equivalent of KEY1-KEY4, or do they stack on the end of that (so if you have 
KEY1, KEY2, and switch1 defined you would get POWER1-POWER3)?

davidelang, 2017-03-09:
On Thu, 9 Mar 2017, Theo Arends wrote:

> Only when you have two or more relays. With one relay and two switches the second switch will control the single relay too.

until you define switchtopic, at which point the two switches will produce 
different codes.

gotcha

Thanks

arendst, 2017-03-09:
No key1-key4 are the sonoff buttons connected to their respective GPIOs. Switches are user supplied items connected to GPIO_USER and numbered from 1 to 4.

arendst, 2017-03-09:
Pfew...

davidelang, 2017-03-09:
On Thu, 9 Mar 2017, Theo Arends wrote:

> No key1-key4 are the sonoff buttons connected to their respective GPIOs. 
> Switches are user supplied items connected to GPIO_USER and numbered from 1 to 
> 4.

Ok, just to confirm I understand this correctly, if two switches were connected 
to a 4ch, the result would produce POWER1-POWER6 (1-4 being the built-in 
switches, 5-6 being the added switches).

David Lang

arendst, 2017-03-09:
Nope.
- When switchtopic is 0 the switches would control relay 1 and 2 and you would receive mqt messages like ``stat/topic/POWER1`` and ``POWER2``. 
- If switchtopic is defined they would send mqtt message like ``cmnd/switchtopic/POWER1`` and ``POWER2``.

mihalski, 2017-03-09:
So if there are relays, switchtopic decouples the relays from the switches. And if there are no relays it enables the use of virtual MQTT switches.

Is that about right?

davidelang, 2017-03-09:
ok, so on a 4ch, a switch3 would duplicate key3? (since switchtopic changes the topic for the key# output as well)

arendst, 2017-03-09:
Yep

davidelang, 2017-03-10:
in 4.0.3 you moved a bunch of things from the template only to user configurable, but you still have switch1 vs button1, is there a difference between them? or should one of them go away in a future version?

arendst, 2017-03-10:
Yes there is a big difference between them and they both will stay.
- Buttons are the default keys as provided by Sonoff (1 for basic, up to 4 for 4CH). They have their own single ButtonTopic and Button1 has special code for initializing the Sonoff. I have some kind of control over the functionality of buttons.
- Switches are user defined items up to 4 having their own code and are not supported as direct replacement for Buttons. They are attachments on user requests needing external switches to available GPIOs. Initially the request was for one switch allowing to control the single relay in a sonoff basic. Over time it has become a kind of one size fits all with their own SwitchTopic allowing for sending MQTT messages only without relay control.

closed 2017-03-12 by arendst

davidelang, 2017-03-16:
so if you set buttontopic to abc and switchtopic to def you can get up to 8 different messages from the sonoff?

now that it's possible to set buttons on gpios, would it make sense to remove switches entirely and just use buttons?

arendst, 2017-03-16:
What I've tried to explain is that Buttons are there by default build in the hardware. Button1 addresses relay1.

People wanted to control the relay also with an external button/switch without physical cabling to the the original button. So I had to come up with code that links switch1 to relay1 too. That's why both have to stay. The fact that Button pop up in the drop down box is that 'other' people wanted to configure buttons on non-sonoff hardware.

It's confusing I know but that's because this software almost becomes a swish army knife...

davidelang, 2017-03-16:
swiss army knives are very useful :-)

it's just that in the process of becoming more general, it sometimes makes sense to revisit the reasons that things are what they are and check that they still make sense in the face of other features that have been added in the meantime.

the need to duplicate the built-in buttons with externally connected switches makes sense. I'm just wondering if it would make more sense to have a way to redefine button1 to a different gpio and simplify things by eliminating the switch/button distinction.

I've also been thinking that with the firmware becoming more general, it may make sense to reorganize the

mihalski, 2017-03-16:
With regards to the COMMANDS being blurted on every power change because there are no relays.. I now have two divergent behaviours from identically configured devices. Both have just one PIR as switch1 on GPIO14. One blurts out the COMMANDS when its state changes and the other does not. Both also the same version of the firmware.

Exhibit A:
```19:33:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:33:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:33:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:33:18", "BME280":{"Temperature":31.8, "Humidity":59.3, "Pressure":1003.0}, "BH1750":{"Illuminance":0}}
19:38:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:38:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:38:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:38:18", "BME280":{"Temperature":31.9, "Humidity":58.8, "Pressure":1003.1}, "BH1750":{"Illuminance":0}}
19:43:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:43:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:43:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:43:18", "BME280":{"Temperature":31.9, "Humidity":58.5, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
19:48:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:48:18", "Uptime":0, "Vcc":2.747, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:48:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:48:18", "BME280":{"Temperature":31.9, "Humidity":58.5, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
19:53:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:53:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:53:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:53:18", "BME280":{"Temperature":31.9, "Humidity":58.2, "Pressure":1003.1}, "BH1750":{"Illuminance":0}}
19:58:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:58:18", "Uptime":0, "Vcc":2.747, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:58:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:58:18", "BME280":{"Temperature":31.9, "Humidity":58.0, "Pressure":1003.3}, "BH1750":{"Illuminance":0}}
20:02:30 MQTT: tele/bedroom/UPTIME = {"Time":"2017-03-16T20:02:30", "Uptime":1}
20:03:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T20:03:18", "Uptime":1, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
20:03:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T20:03:18", "BME280":{"Temperature":31.9, "Humidity":57.8, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
20:03:55 MQTT: stat/bedroom/RESULT = {"POWER":"ON"}
20:03:55 MQTT: stat/bedroom/POWER = ON
20:03:59 MQTT: stat/bedroom/RESULT = {"POWER":"OFF"}
20:03:59 MQTT: stat/bedroom/POWER = OFF
20:04:40 CMND: switchtopic
20:04:40 MQTT: stat/bedroom/RESULT = {"SwitchTopic":"0"}
```

Exhibit B:
```
20:04:54 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:04:54 MQTT: stat/office/POWER = OFF
20:05:00 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:00 MQTT: stat/office/POWER = ON
20:05:00 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:05:00 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:05:00 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
20:05:00 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:03 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:03 MQTT: stat/office/POWER = OFF
20:05:15 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:15 MQTT: stat/office/POWER = ON
20:05:16 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:05:16 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:05:16 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
20:05:16 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:18 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:18 MQTT: stat/office/POWER = OFF
20:05:24 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:24 MQTT: stat/office/POWER = ON
20:05:24 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
[... cut ...]
20:05:24 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
20:05:24 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:27 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:27 MQTT: stat/office/POWER = OFF
20:05:27 CMND: switchtopic
20:05:27 MQTT: stat/office/RESULT = {"SwitchTopic":"0"}
```
Am I missing something?

mihalski, 2017-03-16:
With regards to the COMMANDS being blurted on every power change because there are no relays.. I now have two divergent behaviours from identically configured devices. Both have just one PIR as switch1 on GPIO14. One blurts out the COMMANDS when its state changes and the other does not. Both also the same version of the firmware.

Exhibit A:
```
19:33:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:33:18", "BME280":{"Temperature":31.8, "Humidity":59.3, "Pressure":1003.0}, "BH1750":{"Illuminance":0}}
19:38:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:38:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:38:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:38:18", "BME280":{"Temperature":31.9, "Humidity":58.8, "Pressure":1003.1}, "BH1750":{"Illuminance":0}}
19:43:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:43:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:43:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:43:18", "BME280":{"Temperature":31.9, "Humidity":58.5, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
19:48:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:48:18", "Uptime":0, "Vcc":2.747, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:48:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:48:18", "BME280":{"Temperature":31.9, "Humidity":58.5, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
19:53:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:53:18", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:53:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:53:18", "BME280":{"Temperature":31.9, "Humidity":58.2, "Pressure":1003.1}, "BH1750":{"Illuminance":0}}
19:58:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T19:58:18", "Uptime":0, "Vcc":2.747, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
19:58:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T19:58:18", "BME280":{"Temperature":31.9, "Humidity":58.0, "Pressure":1003.3}, "BH1750":{"Illuminance":0}}
20:02:30 MQTT: tele/bedroom/UPTIME = {"Time":"2017-03-16T20:02:30", "Uptime":1}
20:03:18 MQTT: tele/bedroom/STATE = {"Time":"2017-03-16T20:03:18", "Uptime":1, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":84}}
20:03:18 MQTT: tele/bedroom/SENSOR = {"Time":"2017-03-16T20:03:18", "BME280":{"Temperature":31.9, "Humidity":57.8, "Pressure":1003.2}, "BH1750":{"Illuminance":0}}
20:03:55 MQTT: stat/bedroom/RESULT = {"POWER":"ON"}
20:03:55 MQTT: stat/bedroom/POWER = ON
20:03:59 MQTT: stat/bedroom/RESULT = {"POWER":"OFF"}
20:03:59 MQTT: stat/bedroom/POWER = OFF
20:04:40 CMND: switchtopic
20:04:40 MQTT: stat/bedroom/RESULT = {"SwitchTopic":"0"}
```
Exhibit B:
```
20:04:54 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:04:54 MQTT: stat/office/POWER = OFF
20:05:00 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:00 MQTT: stat/office/POWER = ON
20:05:00 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:05:00 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:05:00 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
20:05:00 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:03 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:03 MQTT: stat/office/POWER = OFF
20:05:15 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:15 MQTT: stat/office/POWER = ON
20:05:16 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:05:16 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
20:05:16 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
20:05:16 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:18 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:18 MQTT: stat/office/POWER = OFF
20:05:24 MQTT: stat/office/RESULT = {"POWER":"ON"}
20:05:24 MQTT: stat/office/POWER = ON
20:05:24 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
20:05:24 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedStat [... cut ...] ode, I2CScan"}
20:05:24 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
20:05:27 MQTT: stat/office/RESULT = {"POWER":"OFF"}
20:05:27 MQTT: stat/office/POWER = OFF
20:05:27 CMND: switchtopic
20:05:27 MQTT: stat/office/RESULT = {"SwitchTopic":"0"}
```
Am I missing something?

EDIT: Thought I had noticed something different but I was wrong so I'm putting it back after deleting it.

davidelang, 2017-03-16:
one thing is that it looks like these are different types of modules (the first is reporting sensor data)

so the problem is that when it gets turned on, you get the output of commands

mihalski, 2017-03-16:
The other one has the same sensor modules. The logs had been going long enough (and with enough additional COMMANDS) to push those telemetry details out.

Here it is again (I've increased the time delay and switched to a repeatable trigger as I'm experimenting with the use case of this particular PIR).
```
21:13:58 MQTT: tele/office/STATE = {"Time":"2017-03-16T21:13:58", "Uptime":2, "Vcc":2.714, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":50}}
21:13:58 MQTT: tele/office/SENSOR = {"Time":"2017-03-16T21:13:58", "BME280":{"Temperature":25.9, "Humidity":34.4, "Pressure":1004.1}, "BH1750":{"Illuminance":3}}
21:17:08 MQTT: stat/office/RESULT = {"POWER":"OFF"}
21:17:08 MQTT: stat/office/POWER = OFF
21:18:09 MQTT: stat/office/RESULT = {"POWER":"ON"}
21:18:09 MQTT: stat/office/POWER = ON
21:18:09 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
21:18:09 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
21:18:09 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
21:18:09 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
21:18:58 MQTT: tele/office/STATE = {"Time":"2017-03-16T21:18:58", "Uptime":2, "Vcc":2.714, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":52}}
21:18:58 MQTT: tele/office/SENSOR = {"Time":"2017-03-16T21:18:58", "BME280":{"Temperature":26.8, "Humidity":35.6, "Pressure":1004.1}, "BH1750":{"Illuminance":3}}
21:20:00 MQTT: stat/office/RESULT = {"POWER":"OFF"}
21:20:00 MQTT: stat/office/POWER = OFF
21:20:11 MQTT: stat/office/RESULT = {"POWER":"ON"}
21:20:11 MQTT: stat/office/POWER = ON
21:20:11 MQTT: stat/office/COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
21:20:11 MQTT: stat/office/COMMANDS2 = {"Commands2":"Mqtt, MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, GroupTopic, Units, Timezone, LedState, LedPower, TelePeriod"}
21:20:11 MQTT: stat/office/COMMANDS3 = {"Commands3":"Power1, Power2, Light1 Light2, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation, SwitchMode, I2CScan"}
21:20:11 MQTT: stat/office/COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}
```
As you can see, the same sensors.

davidelang, 2017-03-16:
Ok, we need to see all the mqtt commands that go to these devices. I'd bet that 
there is something else being sent to the one that is generating the commands 
output (like a dimmer command or something like that)

set logging to more debugging (4) and look at the output

mihalski, 2017-03-16:
YES! I guess this is something that worked with older firmware. WAY back when I had set up a script to automatically toggle a light when motion was detected. It would respond with:
```
21:42:51 MQTT: stat/office/RESULT = {"POWER":"ON"}
21:42:51 MQTT: stat/office/POWER = ON
21:42:51 RSLT: Receive topic cmnd/office/lightpack, data size 2, data on
21:42:51 RSLT: DataCb Topic office, Group 0, Index 1, Type LIGHTPACK, Data on (ON)
```
I can't remember why I did this anymore.. I know there was a reason.

Interesting that the exact same symptom occurs when you have multiple switches not linked to relays. Because in that situation I don't have any corresponding logic running outside.

I'll disable the response and maybe that will clear up why I wanted/needed it in the first place.

davidelang, 2017-03-16:
I would guess that with multiple switches and no relays, it's trying to internally generate the cmnd to turn on the relay and finding there is no such relay.

mihalski, 2017-03-16:
If so that sounds like with the intent of it being used for just that purpose this would be something that could be considered a bug. i.e.. when you have multiple switches, no relays and switchtopic is not 0, avoid this branch of the code.

davidelang, 2017-03-16:
That sort of configuration isn't tested that well, arendst is only testing sonoff devices, mostly without external switches/relays. It's up to the rest of us to test those configs and track down the oddities. :-)

mihalski, 2017-03-16:
Oh, I'm all about tracking down bugs.. I find bugs in ALL software.. Going back to getting free versions of MS Win98 for finding bugs in that.

I don't think the onus is on fixing non Sonoff bugs is on Theo at all.. As for providing solutions.. I just don't have the resources. But I'd be happy to help in any way I can. So I'm your man for any sort of testing..

My next focus is on trying to debug PIR behaviour to figure out whether it's a Sonoff problem, a Wemos problem, a power problem, a quality problem.. or something else. It's seems so random.

kisseler, 2017-06-22:
First of all Thanks, Theo, for your wonderful work!

I am trying for two days now to achieve the following procedure:
I use a Sonoff Basic module to switch a main light. Setting the GPIO14 to Switch1 allows to plug in a wall switch. That works fine. With GPIO3 I want to switch another Sonoff (S20) which is placed in the corner of the room. So I set GPIO3 to Switch2.

What worked at the beginning: I set switchtopic1 to the local Sonoff and switchtopic2 to the S20. That does not work anymore. As soon I set switchTopic2 some value, all switchTopic gain that value. I really don't know what is going wrong here, because at the beginning that worked fine.

And most important: What I was not able to achieve is: I would like the Switch at GPIO3 (Switch2) not to send the command "cmnd/<switchtopic2>/POWER2:TOGGLE", because I receive a "command:unknown" message by the S20. Can I influence the message sent by Switch2 to be "cmnd/<switchtopic2>/POWER:TOGGLE" (without number of the switch)?

Thank you very much for your help!

Kedryn, 2018-07-20:
I'm using Tasmota 5.12 on a Nodemcu board, where i've attached few sensors that work via I2c, and two sensor that works as switches, one IR and one Radar. The Nodemcu has no relays.

I tried to follow instructions i've read here, to have 2 different MQTT messages for those sensors, but it seems i can't manage to get it working.

I've configured Tasmota to use Switch1 and Switch2 for those IR and Radar sensors.
Then i issued a SwitchMode1 1 and SwitchMode2 1

01:32:16 MQT: homeassistant/garage_sensors/POWER = ON
01:32:18 MQT: homeassistant/garage_sensors/RESULT = {"POWER":"OFF"}
01:32:18 MQT: homeassistant/garage_sensors/POWER = OFF
01:32:19 MQT: homeassistant/garage_sensors/RESULT = {"POWER":"ON"}

No changes in POWERn

After that i tried Switchtopic1 IR and Switchtopic2 Radar

01:33:20 MQT: homeassistant/Radar/RESULT = {"POWER":"OFF"}
01:33:20 MQT: homeassistant/Radar/POWER = OFF

Still no POWERn

What i'm missing?

mattpurnell, 2018-08-03:
I am also having no luck getting POWERn..! 

SwitchTopic(n) seems to have no difference to just simply SwitchTopic

Kedryn, 2018-08-03:
Thank to Drrzs on Discord, the problem got solved.
No need for SwitchTopic

1) You need Tasmota >5.12 (i had 5.12 and had to upgrade)
2) You do a "SetOption26 1" command on tasmota
3) Even if you dont have a relay on your node, use a spare pin and configure one relay (virtual). I configured 2 to be sure
4) Configure 2 Switch 
You will have POWER1 and POWER2 events.

My node is now working perfectly. Oh.. yeah, not so perfectly because i get false positives from IR sensor, but that is another problem.

casperse, 2019-10-27:
I just got my first multi switch (Sonoff 4CHPro/Tasmota 6.6.0.21) and I am struggling to set the "PowerOnState" for each switch?

What I am trying to do is having switch 1,2,3 as PowerOnState 1 and switch 4 as PowerOnState 0
Is this at all possible?

Sofar this dosent work:
`16:33:04 MQT: stat/Sonoff_4CHPRO/RESULT = {"PowerOnState":1}
16:33:12 CMD: poweronstate2 1
16:33:12 MQT: stat/Sonoff_4CHPRO/RESULT = {"PowerOnState":1}
16:33:18 CMD: poweronstate3 1
16:33:18 MQT: stat/Sonoff_4CHPRO/RESULT = {"PowerOnState":1}
16:33:25 CMD: poweronstate4 0
16:33:25 MQT: stat/Sonoff_4CHPRO/RESULT = {"PowerOnState":0}`

It gives:
`16:44:16 MQT: tele/Sonoff_4CHPRO/LWT = Online (retained)
16:44:16 MQT: cmnd/Sonoff_4CHPRO/POWER = 
16:44:16 MQT: tele/Sonoff_4CHPRO/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.6.0.21(0ebec1f-sonoff)","FallbackTopic":"cmnd/Sonoff_4CHPRO_fb/","GroupTopic":"sonoffs"}
16:44:16 MQT: tele/Sonoff_4CHPRO/INFO2 = {"WebServerMode":"Admin","Hostname":"Sonoff_4CHPRO","IPAddress":"192.168.0.63"}
16:44:16 MQT: tele/Sonoff_4CHPRO/INFO3 = {"RestartReason":"Power on"}
16:44:16 MQT: stat/Sonoff_4CHPRO/RESULT = {"POWER1":"OFF"}
16:44:16 MQT: stat/Sonoff_4CHPRO/POWER1 = OFF
16:44:16 MQT: stat/Sonoff_4CHPRO/RESULT = {"POWER2":"OFF"}
16:44:16 MQT: stat/Sonoff_4CHPRO/POWER2 = OFF
16:44:16 MQT: stat/Sonoff_4CHPRO/RESULT = {"POWER3":"OFF"}
16:44:16 MQT: stat/Sonoff_4CHPRO/POWER3 = OFF
16:44:16 MQT: stat/Sonoff_4CHPRO/RESULT = {"POWER4":"OFF"}
16:44:16 MQT: stat/Sonoff_4CHPRO/POWER4 = OFF
16:44:18 MQT: stat/Sonoff_4CHPRO/RESULT = {"POWER4":"OFF"}
16:44:18 MQT: stat/Sonoff_4CHPRO/POWER4 = OFF
16:44:24 MQT: tele/Sonoff_4CHPRO/STATE = {"Time":"2019-10-27T16:44:24","Uptime":"0T00:00:16","UptimeSec":16,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"speedzone","BSSId":"F2:9F:C2:27:D3:55","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:05"}}`

Any and all help would be highly appriciated Thanks :-)

meingraham, 2019-10-27:
Wow - you wen't back to the dark ages to find an issue that was questionably related to your request. 

```
Rule1 ON System#Boot DO Backlog Power1 1; Power2 1; Power3 1; Power4 0 ENDON
Rule1 1
```

Please address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

casperse, 2019-10-28:
Sorry I didn't see it was from 2017 - Anyway thanks for trying to help!
I have tried that and it starts on on all 4 after a power cut and first after some seconds it will turn of channel 4 (My hope was that it would start OFF with no delay, like the others in state ON)

meingraham, 2019-10-29:
Set poweronstate so they boot off.

casperse, 2019-10-29:
Yes of course! THANKS!