arendst/Tasmota #90: PowerOn state not working as expected.

radiohe4d, 2017-02-21:
When the Sonoff loses power and starts up again i would expect that when the APP_POWERON_STATE state is set to "0" (OFF at power on) that the device would start and remain off. But after a few seconds it switches ON.

The same happens, only in reverse if the APP_POWERON_STATE state is set to "1" (ON at power on). 
The device powers up and is immediately turned on but after a few seconds it turns itself OFF. 

I have tried setting APP_POWERON_STATE = 3 but get similar unpredictable results.

Im guessing it could be because its connecting to the MQTT server and reading the previous state saved on the server. But even if i turn the device off, so the MQTT server state is OFF, and i power cycle the device which has the APP_POWERON_STATE = 0 the sonoff starts as off but turns itself ON anyway. 

Am i missing something? 
Help would be greatly appreciated as having all 8 devices turning on at 5AM because the electric has reset is not an ideal situation :P Cheers!

davidelang, 2017-02-25:
the devices do not fetch state from the MQTT server, they just connect to it and register to listen for things and then push status info to the MQTT server. If the MQTT server (or something else listening) sees them come up and that their state is not what the central server wants, it could send a command to switch the sonoff.

what are you using as your central server?

you may want to watch the MQTT messages to see exactly what's happening. I have an app on my phone (mqtt dashboard) that lets me subscribe to watch various events (including '#' which shows all messages), you can then see what messages are sent.

radiohe4d, 2017-03-03:
Aww you've made me realise that its possibly Home Assistant (HA) that is setting the state from what it thinks the last state was. 
If this is the case, any ideas how i would ensure that HA only reads the current state of the Sonoff and when the device comes back online, does not try to force the state to what it thinks it should be? 
That way letting the Sonoff always govern what its power on state should be, and continue to be until i interact with it.
Cheers.

closed 2017-03-08 by arendst