arendst/Tasmota #92: Wemo discovery not working with pywemo (HomeAssistant)

aneisch, 2017-02-21:
This issue might be due to web server port. Unsure at this time.

altelch, 2017-02-21:
I think this is related to the very specific way of Echo detecting the Wemo device. The code currently only accepts the echo specific SSDP request. That's probably the problem. In case of Home Assistant I'd use MQTT to controll the Sonoff. The setup is documented in the Wiki.

aneisch, 2017-02-21:
I agree. I actually have some sonoff compatible code that works with Echo/HomeAssistant, but it would be nice if this project was also compliant. Up to the author if they want this to be a real issue or not.

altelch, 2017-02-22:
Please uncomment line 207 in xdrv_wemohue.ino to get the SSDP incoming request when Home Assistant searches for the WeMo. This is what has do be added in line 209 as another valid search request (like in the HUE Emulation a few lines below).

aneisch, 2017-02-22:
```
18:20:13 M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5
```

altelch, 2017-02-22:
Please try: https://github.com/altelch/Sonoff-Tasmota.git

aneisch, 2017-02-22:
Unfortunately no change. I see the device being queried and it sending the response, but it does not show up properly. A python console running a pywemo `devices = pywemo.discover_devices()` does not yield to any results either. That is essentially what Home Assistant is doing, so no surprise there. 

https://github.com/pavoni/pywemo

altelch, 2017-02-22:
Ok, had to change to upnp:rootdevice then msearch answer is sent. After that the pywemo code requires the ST to be set to upnp:rootdevice and not urn:Belkin:device** (changed this in pywemo ssdp.py to test). After that the device is found and pywemo fails when parsing setup.xml. I can't find the error in the xml syntax and I can't change the ST in the MSEARCH response as it is required for ECHO support. So without changing the UDP handler to differentiate the two search types and send the fitting response there will be no autodiscover. And after that I don't know what should be wrong with the setup.xml or what pywemo expects. Won't follow that path any longer as it requires more work than expected and I don't have any use of if. Configured my Home assistant to use MQTT with the sonoffs. Works great.

Sorry,

    Heiko.

aneisch, 2017-02-22:
Sorry to hear, but thank you for looking into this. The auto discovery "issue" isn't a big deal for me personally, but I thought those not as familiar with MQTT could appreciate an easier setup. 

For what it's worth, if anyone is to pursue this in the future, a working setup.xml for pywemo is:

```
  String body = 
  "<?xml version=\"1.0\"?>\r\n"
  "<root xmlns=\"urn:Belkin:device-1-0\">\r\n"
  "<specVersion>\r\n"
    "<major>1</major>\r\n"
    "<minor>0</minor>\r\n"
        "</specVersion>\r\n"
        "<device>\r\n"
          "<deviceType>urn:Belkin:device:controllee:1</deviceType>\r\n"
          "<friendlyName>" + friendlyName + "</friendlyName>\r\n"
              "<manufacturer>Belkin International Inc.</manufacturer>\r\n"
              "<manufacturerURL>http://www.belkin.com</manufacturerURL>\r\n"
              "<modelDescription>Belkin Plugin Socket 1.0</modelDescription>\r\n"
              "<modelName>Socket</modelName>\r\n"
              "<modelNumber>1.0</modelNumber>\r\n"
              "<UDN>uuid:Socket-1_0-" + uuid + "</UDN>\r\n"
              "<modelURL>http://www.belkin.com/plugin/</modelURL>\r\n"
            "<serialNumber>" + serialNumber + "</serialNumber>\r\n"
            "<serviceList>\r\n"
              "<service>\r\n"
                "<serviceType>urn:Belkin:service:basicevent:1</serviceType>\r\n"
                "<serviceId>urn:Belkin:serviceId:basicevent1</serviceId>\r\n"
                "<controlURL>/upnp/control/basicevent1</controlURL>\r\n"
                "<eventSubURL>/upnp/event/basicevent1</eventSubURL>\r\n"
                "<SCPDURL>/eventservice.xml</SCPDURL>\r\n"
              "</service>\r\n"
            "</serviceList>\r\n"
          "</device>\r\n"
        "</root>\r\n"
        "\r\n";
```

mp3llll, 2017-03-10:
so, anesich, can i use your code in order to use iTead Sonoff Touch and ha?

what is the difrent from your code to this?

davidelang, 2017-03-27:
Is this still a problem? or has it been solved?

0n3-70uch, 2017-04-12:
Hello,

normally my home server is working with Domoticz but I want to discover a little bit. So I installed Home Assistant and want to add the Sonoff switch via the Wemo emulation. But the HA server cannont find any Sonoff device (running Tasmota 4.1.3 and Wemo emulation). 

I tried different firmware versions with different setup.xml configurations... without any working auto detection.

Is there anyone with a solution :-).
Greetings
Fabi

altelch, 2017-04-13:
Hi,

no, the Wemo Emulation is just the minimal subset needed to get Amazon 
Echo working, nothing more. If you use HomeAssistant use MQTT or for 
Domoticz the "native" support.

Greetings,

      Heiko.

0n3-70uch, 2017-04-13:
Hello Heiko,

Ok so the same like with Domoticz... I need to use a MQTT server :). I was searching for a plug and play solution but I think at the moment it's only working with original Wemo or Hue products...

Thanks for your help
Fabi

davidelang, 2017-04-23:
This seems like the discussion has ended, so I'm closing it. If you disagree, please re-open it.

closed 2017-04-23 by davidelang