arendst/Tasmota #93: Request: Root Topic setting

KonstantinMastak, 2017-02-21:
Hi!

I tried to connect Sonoff-Tasmota to several private and cloud MQTT brokers and I face the problem that they usually require that client subscribes and publishes to a certain subtopic that is root subtopic for this client. I.e. the root of MQTT broker for a given user is not /, but, say, UserName/ or ClientID/. So I need to specify Root Topic somewhere in Sonoff settings. 

As of now you have fixed root topics like cmnd/ and stat/, and this prevents me to use some MQTT services. I need an ability to specify Root Topic for device, so MQTT path looks like RootTopic/cmnd/sonoff/power.

Can you please do this?

arendst, 2017-02-22:
These fixed root topics are totally under control by you!

Just change the values of SUB_PREFIX, PUB_PREFIX and PUB_PREFIX2 in ``user_config.h`` to your likings. So changing ``cmnd`` to ``clientid/cmnd`` should work just fine.

KonstantinMastak, 2017-02-22:
Theo, you are absolutely right, thank you!

But can you please bring these parameters SUB_PREFIX, PUB_PREFIX and PUB_PREFIX2 to web config interface, so they can be saved and loaded to cfg just like TOPIC and other parameters?

closed 2017-03-12 by arendst