arendst/Tasmota #95: Open access point when WiFi  router down

PaulHam211, 2017-02-22:
I currently have 2 x Sonoff basic and 1 x Sonoff POW running on my network.
Last night I my WiFi router was playing up and it went down a few times. The Sonoff POW started to broadcast its own open access point (the 2 Sonoff basics didn't). I connected to the access point and saw that it brought up the WiFi SSID configuration page (which is great if you need to reconfigure the SSID).

I cant help but think this may be vulnerable to outsiders re configuring the POW rendering it unusable without intervention.

I know there was an issue/discussion about password protecting the web interface, and it was suggested to just disable the web interface but I am not sure that is the best option.

What do you think about an admin password/4 digit pass-code, which is required to enable the saving of settings on the web interface?

arendst, 2017-02-22:
Paul, why not disable the open access point feature? Just change wificonfig to 4 and Sonoff won't go in WebConfig after an Access Point is down; it just retries the other AP if configured.

See Wiki commands about other great features available (like restart on AP loss ;-)

PaulHam211, 2017-02-26:
Thanks. checked this out over the weekend and it is fine, thank you.

Although now I think I have issues due to the newest firmware. I flashed 2 of my older sonoffs (ones which I upgraded the flash chip) and they both kept disconnecting from the router and being unresponsive. I have reflashed (through FTDI) down to 3.9.11 and ill keep an eye to see if things are working reliably like before.

closed 2017-03-08 by arendst