arendst/Tasmota #96: HTU21

Greg2002, 2017-02-22:
I don't know why but HTU21 not work with I2C (BME280 and BH1750 - tested OK).
(sensor tested with Arduino 3.3V Sparkfun Library and works fine).

altelch, 2017-02-23:
Whats the result of command I2Cscan? Is the sensor found?

arendst, 2017-02-23:
My HTU21 works fine. Sometimes it needs a power cycle to reset itself.

Greg2002, 2017-02-26:
I have a Si7021 sensor, not HTU21. With your library it will not work. I have an universal library from GitHub and it works fine with it. Thank you.

altelch, 2017-02-26:
Might be related to the device id. I check for 0x32 as this is the device id returned by the htu21. The Si7021 seems to return 0x15. Please try to change this in the HTU21 driver and test again.

altelch, 2017-02-26:
Added code in my fork. Please try: https://github.com/altelch/Sonoff-Tasmota.git
If it works I'll do a pull request.

Greg2002, 2017-02-26:
It works but give wrong data of Humidity. HTU Temperature: 
20.4°C
HTU Humidity: 
-0.7%
BMP Temperature: 
21.2°C
BMP Humidity: 
30.6%
BMP Pressure: 
745.4 mmHg
Illuminance: 
102 lx
This is due to wrong delay (HTU21 is faster). You can check a working library at https://github.com/enjoyneering/HTU21D

altelch, 2017-02-26:
Changed delay to 23ms. Please try again with my fork. If it works I'll see how to "fix" this is a better way.

Greg2002, 2017-02-27:
HTU Temperature: 
21.2°C
HTU Humidity: 
28.2%
BMP Temperature: 
21.5°C
BMP Humidity: 
27.2%
BMP Pressure: 
752.0 mmHg
Illuminance: 
105 lx
OFF

Greg2002, 2017-02-27:
Now it works

altelch, 2017-02-27:
Hi, could you please test again with my current code? Correct name should now be returned in logfile and the conversion times are set per sensor type.

Greg2002, 2017-02-27:
{"Time":"2017-02-27T15:28:30", "SI7021":{"Temperature":22.0, "Humidity":27.3}, "BME280":{"Temperature":21.4, "Humidity":27.4, "Pressure":755.0}, "BH1750":{"Illuminance":10}}

Greg2002, 2017-02-27:
But web server returns HTU
HTU Temperature: 
21.8°C
HTU Humidity: 
27.4%
BMP Temperature: 
21.7°C
BMP Humidity: 
26.8%
BMP Pressure: 
755.0 mmHg
Illuminance: 
11 lx
OFF

altelch, 2017-02-27:
Ok, web output should be fixed now. Please try again.

Greg2002, 2017-02-27:
OK
SI7021 Temperature: 
21.8°C
SI7021 Humidity: 
27.2%
BMP Temperature: 
21.9°C
BMP Humidity: 
26.6%
BMP Pressure: 
755.0 mmH

Greg2002, 2017-02-27:
You won!

altelch, 2017-02-27:
Ok, will do pull request then.

Greg2002, 2017-02-27:
OK

Greg2002, 2017-02-27:
Today I received TSL2561 for testing instead of  BH1750...

arendst, 2017-03-01:
SI70xx is supported in version 3.9.22

closed 2017-03-01 by arendst