arendst/Tasmota #97: Request: SHT11 support

bluebird11, 2017-02-23:
SHT11 is a popular temperature and relative humidity sensor featuring better accuracy than DHT22. Any chance you could add support for it?

There already exist some implementations for ESP8266, e.g. in ESPEasy and in espweather.

davidelang, 2017-02-25:
it's probably useful to point at an existing implementation (assuming the licenses are compatible)

bluebird11, 2017-02-25:
The implementation can be found in the aforementioned projects (both GNU license):

https://github.com/MalteP/espweather/blob/master/firmware/user/sht1x.c
https://github.com/letscontrolit/ESPEasy/blob/master/_P031_SHT1X.ino

I would be happy to be a beta tester.

arendst, 2017-02-25:
I won't support the SHT11. It's using two wires and needs special code to drive it. 

With two wires I support I2C and the excellent HTU21 temp/hum sensor.

closed 2017-03-08 by arendst