arendst/Tasmota #98: cant flash sonoff 4ch

rayzor666, 2017-02-23:
hi im tryoing to flash a 4ch with no luck.. im getting error "sonoff.ino:144:75: fatal error: ArduinoJson.h: No such file or directory" with ArduinoIDE, with Atom it will flash but the device is useless... can someone pls help with any method of flashing... thanks

lucwuyts, 2017-02-24:
1. in ArduinoIDE - Sketch - Include Library - Manage Libraries -> add the arduinoJson library.
2. Set the board to "Generic ESP8285 module"
3. Flash size:  1M (64K SPIFFS)
4. And the correct comm port.

This should do it.

PS: During flashing i could not power the board from the serial interface, so i power it now with a 3V battery during flash.

rayzor666, 2017-02-24:
thanks for that now im getting "xdrv_ir_send.ino:31:29: fatal error: IRremoteESP8266.h: No such file or directory" any idea how to fix this

lucwuyts, 2017-02-24:
This is explained on this page:
https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite

rayzor666, 2017-02-25:
Thank you very much for your help.... I took a step back, read the wiki properly and started again... All working great

closed 2017-02-25 by rayzor666