arendst/Tasmota #100: Flickering when using switchmode toggle

Splitter, 2017-02-25:
I have one sonoff, running an older version of tasmota(before the name/repository change), and it works completely fine. I have it set up with an external toggle switch and it runs fine. Switching the external toggle switch off or on changes the state of the sonoff to the opposite state. 

I purchased 3 more sonoffs and all three have an issue of flickering. The relay is kicking off and on randomly if the toggle switch is in the off position. If I leave the toggle switch in the ON position and control the sonoff from mqtt or the factory button then there is no issue, which is all the more confusing. It only happens when I have switch mode set to toggle and the external toggle switch in the off position. 

 I have tested all three new sonoffs with both the older version of tasmota that is running on the sonoff I originally purchased and the latest version here. The issue persists with both versions. 

Any idea what could be the cause of this issue?

arendst, 2017-02-25:
What gpio are you using?  Sounds like a floating input when the switch is open although I assign the input with pullup...

Splitter, 2017-02-25:
Its a basic wifi sonoff. Gpio14, the 5th pin away from the factory button where you solder on the headers. What do you mean by floating input?

davidelang, 2017-02-25:
If an input is completely disconnected from anything, it is considered floating, 
and will randomly return 1 and 0

the solution to this is to connect a resistor between that input and voltage 
(pull-up resister) or ground (pull-down resister) and then hook the switch to 
the other voltage (so if you us a pull-up resister, the switch would go between 
the pin and ground)

This way when the switch is not set, the resistor 'pulls' the pin to a defined 
value.

in these chips, they have made it easy and you can configure a pin to be 
floating, or pulled up/down internally to the chip. @arendst is saying that he 
thought the software configured the pin in the 'pull-up' configuration, but your 
symptoms are similar to what would happen if you didn't have this.

David Lang
k

Splitter, 2017-02-26:
So it seems that is the problem. I soldered in a 10k pull up resistor on one to test and it is functioning perfectly, like the original I had. Seeing as I attempted to use the same version of tasmota on these as the working one I can be confident it's not a software issue, so thanks for the help.

Would you have an idea as to why these would act different then the other? Why the internal pull up resistor would fail on all three? Just a bad batch of esps?

closed 2017-02-28 by Splitter

DNA99, 2017-07-08:
Hi, I have the same problem when I am trying to use the sonoff basic to connect to a 3 way switch (cables total length ~15-20meters.). It seems that thay the circuit is closed, the sonoff is not able to consistantly detect the closed circuit and the lights go on and off randomly. Could you show me a pic of how you solder this pull up resistor?

davidelang, 2017-07-08:
you solder the resistor between 3.3v and the GPIO pin.

iandronowicz, 2017-12-23:
How long can the cables from the board and the external switch can be before starting to notice the cable noise? I know that it would depend on the type of cable being use, and the electrónica radiation of the place, etc. But just to know in the worst case scenario what would it be? 2 meters? 5 meters?

Thanks!

davidelang, 2017-12-23:
too many variables, some people will have horrible problems with just a few 
inches of wire, others will have no problems with many feet of wire.