arendst/Tasmota #103: OTA upgrade from Sonoff-MQTT-OTA-Arduino

davidelang, 2017-02-25:
since these are both Arduino based, is it possible to do an OTA upgrade from the old version to the new?

I would expect that the user would then need to login to the device to set it's type and that sort of thing, but it would be nice to not have to pull all the devices and open them up to switch them :-)

arendst, 2017-02-25:
Yep, should work. Try one and enjoy.

davidelang, 2017-02-25:
I confirmed that it works on a TH16

davidelang, 2017-02-26:
it works for the ESP8266 systems, but not on the ESP32 system (wion)

on that system, I compiled the original image with 4M flash (3m spiffs), but when I try to flash a new image I get the error:

stat/sonoff/UPGRADE = Failed Update error: ERROR[8]: Flash config wrong real: 1048576 IDE: 4194304

if I try to flash an image compiled for 1M (64K spifs) or:

stat/sonoff/UPGRADE = Failed bin for wrong flash size

if I try to flash an image compiled for 4m(3m spifs)

status shows:

stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":440, "Free":580, "Heap":27, "SpiffsStart":1024, "SpiffsSize":3052, "FlashSize":1024, "ProgramFlashSize":4096, "FlashChipMode",2}}

any suggestions as to what to try?

davidelang, 2017-02-26:
never mind, that item started out with tasmota, so a separate issue if I still have trouble.

closed 2017-02-26 by davidelang