arendst/Tasmota #104: Could someone please explain switchtopic to me?

mihalski, 2017-02-25:
What I am TRYING to do now that there is functionality for multiple switches, is to connect 2 PIR sensors to different inputs and have them send unique topics when they go on/off.

It doesn't look like you can set the topic for a specific switch..

arendst, 2017-02-25:
You set one switchtopic for all four switches which will result in messages like cmnd/switchtopic/POWER1 TOGGLE/ON/OFF to cmnd/switchtopic/POWER4.

Every switch has it's own switchmode setting.

mihalski, 2017-02-25:
Thank you. That makes sense.

However I don't seem to be getting any integer appended to POWER despite having 2 switches defined like so:

Module type: 17 User Test
GPIO2: 9 Switch 1
GPIO3: 0 None
GPIO4: 0 None
GPIO5: 0 None 
GPIO14: 10 Switch2

Have I neglected to do something or should I be seeing POWER1 and POWER2 with it configured like that?

davidelang, 2017-02-25:
see #88 this functionality was just added a couple days ago, make sure you are 
using the most current version

mihalski, 2017-02-25:
I am using the most current version and the functionality seems to be missing.

mihalski, 2017-02-26:
Anyone have any suggestions? I'm stuck at this point.

closed 2017-03-12 by arendst