arendst/Tasmota #105: Feature request: Display switch state in web interface

mihalski, 2017-02-25:
If it would be possible to display the switch state on the main page much in the same manner as the sensors (at least optionally) that would be great.

davidelang, 2017-02-25:
given that most switches are momentary, this may not help as much as you want.

you can select the console and it will show you all the messages generated, 
including those as you push a switch.

David Lang
k

mihalski, 2017-02-25:
PIT sensors however are either HIGH or LOW so it would certainly help with those.

closed 2017-03-08 by arendst

badabing2005, 2018-11-18:
A magnetic switch to indicate if a Garage is open or closed would be another good example.
One magnetic switch and one reed relay is all that's needed to use Tasmota as an standalone Garage Door opener without the need for MQTT or Home Assistant, only if the state of the switch was indicated on the main page.

Let's hope one day this would be possible.

Thanks
BB

KDMcMullan, 2019-03-10:
Ah. I was about to raise this as a feature request.

> given that most switches are momentary,

I'm pretty sure that "momentary" is resolved as a "button" rather than a switch. I would hope that switch statuses could be displayed on the web interface, but do agree that button statuses might be fairly pointless (unless the user had a mechanically held button, I suppose).

I realise the author is a busy man, but I was wondering if this was "cannot fix" or a "will not fix"? I'd be prepared to have a stab at a PR myself with a gentle steer in the right direction! Is it a complete non-starter?

Many thanks for a wonderful piece of work.

Ptrvh, 2020-10-14:
This would indeed be a great feature, certainly for the garage door opener. I also use this as stand alone and then this is a great feature.

victorclaessen, 2020-11-11:
I agree that this would be a useful addition.

madduck, 2023-04-30:
Just mentioning this here: `SetOption114 0` will make it such that switches control the corresponding relays. With `SetOption114 1`, you need a rule like `on Switch1#state do Power1 %state% endon` to mirror the switch state onto the relay.

It's unfortunate that the "toggle" button cannot be removed, i.e. the relay made read-only from the point of view of the Web UI.