arendst/Tasmota #107: identify which gpio is default for sensors on th

davidelang, 2017-02-25:
using the th series deivces, the built-in sensor port is gpio14, but once you select the th series, you are presented with three gpios to configure, without any indication as to which is which.

It would be very useful to be able to identify which one is the default sensor port on the th series when it's selected.

arendst, 2017-02-26:
As you know, gpio14 is the default port.

As some want to add extra hardware for gpio04 it is also available for use.  Gpio03 is always available on the serial port.

As I get requests from those only wanting to see easily available gpios while others want to see all gpios I have to make decisions about useability and code size.

Explaining gpios hs no priority for me. You can always edit sonoff_template.h and remove the ones you do not want to see.

davidelang, 2017-02-26:
I found that it was port 14 by looking at the prior repo. I couldn't find the 
info anywhere in the wiki (I may have missed it)

I agree that they should all be shown (in fact, I think there should probably be 
more GPIO_USER entries)

I'm just thinking that there should be an ability to add a comment to a template 
to be displayed on the configuration page.

is it reasonable to add this to the datastructure holding the GPIO defs? or 
would it need to be a separate array like you have for user defined functions 
and their descriptions?

arendst, 2017-02-26:
I'm running out of memory and have to choose between features or descriptions. Currently there is just 4k left...

davidelang, 2017-02-26:
would you accept a patch that was protected by ifdefs?

On Sun, 26 Feb 2017, Theo Arends wrote:

> Date: Sun, 26 Feb 2017 10:49:11 -0800
> From: Theo Arends <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f168c009c03bffa9122e311fcf18c8cae66eb980892cf0000000114cae52
>     792a169ce0c8878dc@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: davidelang <david@lang.hm>, Author <author@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] identify which gpio is default for
>     sensors on th (#107)
> 
> I'm running out of memory and have to choose between features or descriptions. Currently there is just 4k left...
>
>

arendst, 2017-02-26:
give it a try

arendst, 2017-03-09:
Will implement some text to well known GPIOS like **GPIO14** = TH Default in next version.

closed 2017-03-12 by arendst