arendst/Tasmota #111: Delay before executing mqtt command

lucwuyts, 2017-02-26:
I'm using firmware 3.9.21
Sonoff_4CH

i notice a big delay (seconds) between sending the mqtt command, and executing it. (POWER1 ON or OFF)

i've put up a mqtt viewer, and the command is immediately visible after sending it.
It seems to take between 2 and 7 seconds before executing the command

arendst, 2017-02-26:
What happens on the web console with weblog set to 3?

lucwuyts, 2017-02-26:
I will try to make a screen-movie of this...

lucwuyts, 2017-02-27:
I think the MQTT connection is not stable.
Also: when MQTT not connected, the web interface seems slow  (Reboot count stays the same count)

13:19:11 MQTT: Connected
13:19:16 RSLT: tele/sonoff_CV/LWT = Online
13:19:21 RSLT: cmnd/sonoff_CV/POWER = 
13:19:32 RSLT: tele/sonoff_CV/INFO1 = {"Module":"Sonoff 4CH", "Version":"3.9.21", "FallbackTopic":"DVES_82579D", "GroupTopic":"sonoffs"}
13:19:32 RSLT: tele/sonoff_CV/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff_CV-6045", "IPaddress":"10.0.250.72"}
13:19:32 HTTP: Handle root
13:19:34 MQTT: Attempting connection...
13:19:54 MQTT: CONNECT FAILED, rc -4. Retry in 10 seconds
13:19:54 HTTP: Handle info
13:19:54 HTTP: Handle info
13:19:55 RSLT: stat/sonoff_CV/RESULT = {"POWER1":"ON"}
13:19:55 RSLT: stat/sonoff_CV/POWER1 = ON
13:19:55 RSLT: stat/sonoff_CV/RESULT = {"POWER2":"OFF"}
13:19:55 RSLT: stat/sonoff_CV/POWER2 = OFF
13:19:55 RSLT: stat/sonoff_CV/RESULT = {"POWER3":"OFF"}
13:19:55 RSLT: stat/sonoff_CV/POWER3 = OFF
13:19:55 RSLT: stat/sonoff_CV/RESULT = {"POWER4":"OFF"}
13:19:55 RSLT: stat/sonoff_CV/POWER4 = OFF
13:19:56 HTTP: Handle console
13:20:02 RSLT: tele/sonoff_CV/STATE = {"Time":"2017-02-27T13:20:02", "Uptime":0, "POWER1":"ON", "POWER2":"OFF", "POWER3":"OFF", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"A-D-ENET1", "RSSI":90}}
13:20:06 MQTT: Attempting connection...
13:20:30 MQTT: CONNECT FAILED, rc -4. Retry in 10 seconds

davidelang, 2017-02-27:
If the mqtt connection is unstable, the device tries to connect, and while it is 
trying, everything stops (web interface, responding to the local button, etc)

lucwuyts, 2017-03-07:
It seems these problems are (in my case) not related to the firmware, but to the chip crashing and rebooting due to electric problems.   

This sonoff is a 4CH device.   There is 1 test-lamp connected to it.
When the lamp is not connected: the relays respond immediate,
With the lamp connected, nearly always, the device seems to hang when switching on.

This is a 40W lamp.

Last firmware tested : 4.0.1
Info4 message was always   "Blocked Loop"

closed 2017-03-08 by arendst