arendst/Tasmota #112: Any update on Sonoff SC support?

KonstantinMastak, 2017-02-26:
Do you have plans to add Sonoff SC support, and how fast can this happen? Thank you!

jaysuk, 2017-02-26:
Try this in the interim.

https://github.com/xoseperez/sonoffsc

KonstantinMastak, 2017-02-26:
I know about this projects, thank you! Just wanted to have something like "All Sonoff in one firmware", e.i. was looking for a merge between these two projects, to leverage all avaliable features of both.

arendst, 2017-02-26:
It'll take at least a month as I still haven't received my SC

KonstantinMastak, 2017-02-26:
Thank you! It's great that you have it in short plans. Hope you'll receive your SC soon.

khcnz, 2017-03-20:
@arendst have you received your SC? I have one and happy to start looking at it to give you a headstart. There is some good starting information here as well http://tinkerman.cat/sonoff-sc-with-mqtt-and-domoticz-support/

arendst, 2017-03-20:
Nope, still waiting for it to be send so add another month...

arendst, 2017-04-10:
Initial support in next release

closed 2017-04-10 by arendst