arendst/Tasmota #113: SR04 Ultrasonic sensor support?

Killobite, 2017-02-26:
First off, fantastic firmware!  I have installed this firmware on more than 20 Sonoffs and am really appreciating the capabilities and attention to detail!

I do find that I would really like to add the use of a SR04 distance sensor for one of my implementation and am wondering if there may be any plan or ideas on how to do so?

Thanks!

arendst, 2017-03-05:
The SR04 needs 5V. Sonoff can only supply 3V3. Do you expect the SR04 to work reliably on 3V3?

AlexTransit, 2017-03-05:
Sonoff basic - use 2 power
5v - via trancformater (power relay)
3.3v - afrer stabiliser

Killobite, 2017-03-05:
I will actually be using it with a Wemos D Mini and can take the 5V right off the VCC pin.  I have a mock up working with 3.9.21 and just threw the sonar code in the void loop and sent the output to serial.  I started to dig into an "easy" way to publish the distance into MQTT but that's when I thought I would reach out and see if anybody has any plan/code on it. I had to put a little millis delay in to get it to read correctly and only report out if the distant changes by more than 10 inches but it works reliably, I just have to figure out hot to make it pretty for the web interface and MQTT output.

My use case is a garage door control with open/closed (distance) indicator and a sump pump monitor/override with water level (distance) logging.

Here is my simple loop code in case you are interested:
if (millis() >= lMillis + 250) {

      lMillis = millis();
      sDistNew = Ultrasonic.distanceRead(INC);
    
      if (sDistNew > sDist + 10 ){
        sDist = sDistNew;
        Serial.print("Distance in Inches: ");
        // Pass INC as a parameter to get the distance in inches
        Serial.println(sDist);
        //delay(1000);
      }
      else{
          if (sDistNew < sDist - 10 ){
          sDist = sDistNew;
          Serial.print("Distance in Inches: ");
          // Pass INC as a parameter to get the distance in inches
          Serial.println(sDist);
          //delay(1000);
      }
     }
}

f0m3, 2017-03-18:
Thats a cool idea. Id like to use the sonoff dual for my garage door control but id like to know if it is open or closed too...

hamwong, 2017-03-24:
Wish SR04/05 will support in sonoff, power should not be the reason make SR04 not add as SR501 also use 5V

quadflight, 2017-06-13:
Great idea. Did you get it to work?

BernardLaurenzo, 2017-10-22:
The sonoff dev board has 5v 
Im currently running my SR04 and sonoff dev with ESP easy firmware

jetema, 2017-10-25:
[image]

Yes, 5 volts are available on D5 ( zenner diode ) terminals, you can get a few extra mA from this point to supply SR04

Gompka, 2017-10-25:
You can also use this sensor [https://www.bananarobotics.com/shop/US-100-Ultrasonic-Distance-Sensor-Module](url) as it runs on 3.3v

nosaj66au, 2018-01-29:
Hi Guys,
  has this gone any further I required it for rain water tank level

Jason

fragolinux, 2018-04-30:
HC-SR04P can can work at 3.3V, you can recognize it because usually is blue as the other model, but has NO crystal oscillator on the front panel, look at image
[image]

scottchristian, 2018-05-11:
I'm going to throw my request for this as well.
Would be very nifty

Frogmore42, 2018-06-03:
Well you are in luck.
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR04-Ultrasonic-Sensor

ascillato, 2018-06-13:
Now it is implemented. Closing.

closed 2018-06-13 by ascillato2

Gompka, 2018-06-13:
C.f. gtcy

ascillato, 2018-06-13:
?

Gompka, 2018-07-18:
Yea it does, I have been using it for over a year.

rajil, 2018-08-03:
@Gompka I tried to use PingSerial library with US-100 but that did not [work](https://github.com/stoduk/PingSerial/issues/6). Do you have any example code which works on nodemcu?

Gompka, 2018-08-03:
@rajil, did you make sure the jumper on the back of the us-100 is in the correct position?

rajil, 2018-08-04:
@Gompka I was able to get it working after taking off the jumper. Is it possible to hook two ultrasonic sensors on the same ESP8266. In Tasmota 'configure module' 73/74 device type is Trigger/Echo. I assigned these to two devices, but only one device showed up.

nikito7, 2022-10-13:
Can we change trigger time?