arendst/Tasmota #114: MQTT truncated with multiple DS18B20 sensors

ramsaymike, 2017-02-26:
Trying to use multiple DS18B20 sensors with Sonoff Basic  It seems that the MQTT message is truncated at 239 characters for some reason - part way through sensor 3 address.  This is seen in both the console and Mosquitto.  The web interface reports all data (well, up to the 4 I tried anyway) as expected.  Any ideas?

lucwuyts, 2017-02-26:
maybe you need to increment the MQTT_MAX_PACKET_SIZE

arendst, 2017-02-26:
Change in  sonoff.ino line 125 MESSZ to  a higher value. You'll then also have to increase MQTT_MAX_PACKET_SIZE.

ramsaymike, 2017-02-26:
Thanks - MESSZ was the problem!  And it was sitting right there in the comment!!!
BTW, this stuff is a work of art - fantastic job.  I just wish I could use it for all my 8266s.

closed 2017-02-26 by ramsaymike

2delarosa, 2017-02-27:
@ramsaymike, I'm looking to attach DS18B20 Sensors to either the Sonoff Basic or POW and monitor the temperature of my aquarium.  How are you physically attaching the DS18B20?

ramsaymike, 2017-02-27:
With respect to the Sonoff Basic the DS18B20 is attached to the same header
that was used to program the unit for power and the data line going to the
unused pin 5 - GPIO14 - on the header.  Works great with TASMOTA.  I have
tried 4 sensors without any problems - other than increasing MESSZ if you
use MQTT.

2delarosa, 2017-02-27:
Thanks.  Any need for calibration from the software side?

ramsaymike, 2017-02-27:
I did not do any - I use many of these sensors and for the cost, they are
very good.  I sometimes think about creating individual offsets but for my
application (general house stuff) there really is no need.

altelch, 2017-02-27:
The DS18x20 is factory calibrated but readings differ about +/- 0.5°C.

2delarosa, 2017-02-28:
Thanks.  I ordered several sensors to try them out.

lkhaskjdhlkashd, 2017-03-22:
Thank you very much for this great piece of software!

I'm trying to connect a 18B20 temp sensor to a Sonoff Basic, but i cant get a reading.

I connected the sensor to pin 1-4-5, the same pins i used for flashing the sonoff, is this correct?
pin 1 - vcc
pin 4 - gnd
pin 5 - data

the on/off switch works perfect but i cant seem to get any reading from the sensor.
What do i do wrong?

ramsaymike, 2017-03-22:
Looks OK to me.  Make sure you have a pull-up resister (I use 10k) between
data and vcc.

davidelang, 2017-03-22:
a pull-up resister should not be needed (it is done internally to the esp, but 
it doesn't hurt to try it), are you sure you set the gpio for the right sensor 
type in the software?

ramsaymike, 2017-03-22:
For some reason mine would not work without it....