arendst/Tasmota #115: Sonoff Touch not working - Just light flashing after connected to AC

tringler, 2017-02-26:
Hello,

who is currently using a Sonoff Touch? I was able to flash the ESP successfully. At least it looks like:

```
Build options changed, rebuilding all
Sketch uses 464880 bytes (48%) of program storage space. Maximum is 958448 bytes.
Global variables use 45592 bytes (55%) of dynamic memory, leaving 36328 bytes for local variables. Maximum is 81920 bytes.
Uploading 469024 bytes from C:\Users\mail\AppData\Local\Temp\arduino_build_397211/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 17% ]
................................................................................ [ 34% ]
................................................................................ [ 52% ]
................................................................................ [ 69% ]
................................................................................ [ 87% ]
...........................................................                      [ 100% ]
```

If I connect the ESP to AC my ceiling light is just flashing with a delay of I guess 1 second. The ESP is not connecting to my WiFi. It looks like the firmware is not working at all. By the way if I disconnect the ESP from the Sonoff Touch it keeps flashing.

I tried it with 2 different Sonoff Touch and 2 different ESPs to check if a Touch part or a ESP is broken. Same behaviour.

I tried the flashing procedure many times even with ESP8285 platform or ESP8266 with DOUT option activated.
Same behaviour.

By the way, my Sonoff Switches working like a charm 👍 

Who can help me to solve this issue? Do I forget to set an option?

tringler, 2017-03-02:
It went out that it's not an ESP (and flashed image) problem.

closed 2017-03-02 by tringler

ozbfat, 2017-06-08:
Hi,

I have the same issue, how did you solve it? :/

tringler, 2017-06-08:
Take a look here: https://groups.google.com/d/msg/sonoffusers/p49_ekWZe9k/jilNq8o-CwAJ