arendst/Tasmota #116: How to to use GPIO14 with a 5x PUSH button switch?

Rudoxxxx, 2017-02-26:
Hello!
In my home I have big HALL with 5x PUSH button switch

I want to use GPIO14 with a 5 PUSH button switch. Push one time and relay on, push second time and relay off
I install last version ( Sonoff and ElectroDragon by Theo Arends VERSION // 3.9.21) off programm, I change to SWITCH_MODE PUSHBUTTON, in 2 places, but not help.

    Sonoff.ino Line #87:
    #define SWITCH_MODE PUSHBUTTON // TOGGLE, FOLLOW or FOLLOW_INV (the wall switch state)

    user_config.h line #113
    #define SWITCH_MODE PUSHBUTTON // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON or PUSHBUTTON_INV (the wall switch state)

Where I need to change SWITCH_MODE to PUSHBUTTON
Thanks for help and Thanks for this project.

0n3-70uch, 2017-03-01:
Hello,

You can use the user_config.h or better you use the user_config_override.h with this lines:
`//#ifdef
 CFG_HOLDER
//#undef SWITCH_MODE
//#endif
//#define SWITCH_MODE            PUSHBUTTON`

A second option is the configuration via MQTT. For this check the Commands in the wiki pls.

Rudoxxxx, 2017-03-01:
Yes, this all understand and parameter is changed. Thanks!
Next problem and Question:
I will switch relay on/off with MQTT command. I tray to send "ON" 
using command: stat/sonoff/POWER	ON
using command: stat/sonoff/POWER	OFF
 but not working.

This programm code is ready for receiving data from MQTT.
In my big HALL we have 5x PUSHBUTTON switch, and now new system working good, but we need put 2 more WI-FI switch without wiring from stairs(ladders 2nd floor cellar stage).
We need found the way to change relay state 0/1 from MQTT commands in Sonoff module.

Rudoxxxx, 2017-03-01:
I cant find MQTT read command "mqttCallback()" in this programm for reading MQTT incomming messages :(

davidelang, 2017-03-01:
On Wed, 1 Mar 2017, Rudoxxxx wrote:

> I will switch relay on/off with MQTT command. I tray to send "ON"
> using command: stat/sonoff/POWER	ON
> using command: stat/sonoff/POWER	OFF
> but not working.

stat/* is messages from the sonoff reporting a change

cmnd/* is messages to the sonoff telling it to change

so send cmnd/sonoff/POWER ON instead.

David Lang

Rudoxxxx, 2017-03-03:
Thanks!
cmnd/sonoff/POWER ON/off or 1/0 working

I will make sonoff termorelay for heater. Do you have another programm code or I can use this?

Now I need found the way how to get temperature from MQTT command:
tele/sonoff/SENSOR	{"Time":"2017-03-03T22:05:36", "DS18B20":{"Temperature":25.4}, "DHT":{"Temperature":23.2, "Humidity":55.5}}

Majerdomo.ru server have PHP and Java script, but how to do it with easy way?
/DS18B20Temperature    25.4
/DHTTemperature   23.2
/DHTHumidity    55.5

Rudoxxxx, 2017-03-04:
cmnd/sonoff/POWER ON working, but if I need PUSHBUTTON MODE, I need send only relay change signal.
Sombady ideas?

davidelang, 2017-03-16:
I don't understand exactly what you are asking.you can send cmnd/sonoff/power toggle if you just want to switch it and don't care if it switches it on or off (doesn't seem right for a temp controlled device), remember it's possible for it to get switched locally and network issues prevent the status update from getting to you

you can send the status command to get various parts of the status output, that should let you query the temps instead of waiting for the periodic update

Rudoxxxx, 2017-03-16:
THANKS, TOGGLE  is a right command for change relay state, when I push one of my 5 buttons.

Thanks for help

closed 2017-03-16 by Rudoxxxx