arendst/Tasmota #117: help needed: openhab rssi reading

jesusvallejo, 2017-02-27:
I can´t get the rssi on openhab read the jsonpath.
this is my code in sonoff.item:
`String LivingRoom_Leds_RSSI "Living Room Leds: RSSI [%d %%]" (gRSSI){ mqtt="<[broker:tele/sonoff3/STATE:state:JSONPATH($.Wifi.RSSI)]" }`
and this is my sonoff.sitemap:
`Frame label="RSSI Leds" {
   Text item= LivingRoom_Leds_RSSI labels= "RSSI Leds"

       }     
`
may someone let me know if its well coded or if i need any other code.

thank you

hamwong, 2017-02-28:
{ mqtt="<[mosquitto:tele/sonoff3/STATE:state:JSONPATH($.Wifi.RSSI)]" }

jesusvallejo, 2017-02-28:
Sorry , but I can't see the difference , my broker in the mqtt.cfg is named broker.And thats the only thing i can see different.

jesusvallejo, 2017-02-28:
solved, i had to install from paper ui , map and jsonpath, then go to addon.cfg and add both to the transfomation line, and restart openhab , and voila, its working.

closed 2017-02-28 by jesusvallejo

gonzunigad, 2018-05-28:
Thanks. Had the same problem and i solved it installing JSON PATH transformation on Paper UI.