arendst/Tasmota #120: Either combining or differentiating the TELEMETRY topics

mihalski, 2017-02-28:
Is it possible to either have 2 seperate topics for the sensor telemetry and the status telemetry? They currently appear separately under the same topic.

Alternatively having it all combined into one JSON string would work too.

I ask because with Home Assistant and a sensor entry such as:
```
  - platform: mqtt
    name: "Master Bedroom Temp"
    state_topic: "tele/sonoff_th_master_bedroom/TELEMETRY"
    unit_of_measurement: "°C"
    qos: 1
    value_template: '{{ value_json.DHT.Temperature }}'
```
results in repeated errors:
```
17-02-28 18:41:32 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'DHT' (value: {"Time":"2017-02-28T18:41:32", "Uptime":21, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":92}}, template: {{ value_json.DHT.Temperature }})
```
Is this something that it is possible to do now or would it have to be added?

Thanks in advance.

Regards,
Michal

arendst, 2017-02-28:
You might want to upgrade to at least version v3.9.11 as from that release all JSON messages have unique topics.

mihalski, 2017-02-28:
Oh good lord!

It seems that I completed missed upgrading that sensor :/ All the others are running the current version.

Apologies for being an idiot.

Regards,
Michal

closed 2017-02-28 by arendst