arendst/Tasmota #123: My HomeBridge plugins

MacWyznawca, 2017-02-28:
Hi, I made a few plugins for HomeBridge specifically for Sonoff with this firmware. It may be useful to someone ;-)
https://www.npmjs.com/~macwyznawca

HairBear22, 2017-02-28:
This is incredibly helpful! Thank you for sharing!

MacWyznawca, 2017-03-01:
New wersion for Tasmota 3.9.22 firmware (new Keywords for themperature and humiditi sensors) is ready.

arendst, 2017-03-03:
Do you mind if I link from my wiki to https://www.npmjs.com/~macwyznawca as HomeBridge info?

MacWyznawca, 2017-03-03:
It would be great!
Soon I will add plugin specifically for Sonoff Pow (stores the data in the file with HomeBridge not tiring Sonoff flash memory) 😉

arendst, 2017-03-03:
Added it to the wiki. Feel free to add some more basic info about HomeBridge as my knowledge is NULL

closed 2017-03-05 by arendst

ionciubotaru, 2017-03-06:
Dear MacWyznawca
Can you write few words why you choose HomeBridge instead of Domoticz ?
What is the main difference between them ? 
Which one is the best HomeBridge, Domoticz, or OpenHub ?

MacWyznawca, 2017-03-06:
I am an Apple User since 1991, I do not like a complicated life;-)

Until (hopefully never) Apple does not block access to HomeKit by HomeBridge, I prefer to use native solutions. HomeKit is comfortable, nice, efficient and completely enough for me. In my house everyone uses Apple devices.

The most important (for me): OpenHAB is too complicated to set up for me, and domotich too simple.

MacWyznawca, 2017-03-06:
One more: the HomeBridge can (already) alone to write plugins to my own needs. Eg. Recording historical measurements to a file, or the daily temperature min./max.