arendst/Tasmota #125: Questions about differentiate between switches via both topic and switch mode.

mihalski, 2017-03-01:
Given the following configuration:

"{"GPIO1":0 (None), "GPIO2":9 (Switch1), "GPIO3":0 (None), "GPIO4":6 (I2C SDA), "GPIO5":5 (I2C SCL), "GPIO9":0 (None), "GPIO10":0 (None), "GPIO12":0 (None), "GPIO13":0 (None), "GPIO14":10 (Switch2), "GPIO15":0 (None)}"

and the two following scenarios:

1. Both Switch1 and Switch2 are PIR sensors.
2. Switch1 is a PIR sensor and Switch2 is a momentary button.

Firstly, can I differentiate between the 2 PIR sensors in scenario 1? There is no POWER1/POWER2. Just POWER. Secondly, if there is some way of differentiating, would it be possible to add a feature whereby the momentary switch can be set to switch mode TOGGLE, while the PIR sensor is set to switch mode FOLLOW?

Thank you in advance.

Regards,
Michal

davidelang, 2017-03-01:
On Tue, 28 Feb 2017, mihalski wrote:

> Given the following configuration:
>
> "{"GPIO1":0 (None), "GPIO2":9 (Switch1), "GPIO3":0 (None), "GPIO4":6 (I2C SDA), "GPIO5":5 (I2C SCL), "GPIO9":0 (None), "GPIO10":0 (None), "GPIO12":0 (None), [...]

my understanding is that switches only send messages to your mqtt server, so the 
smarts have to be there.

this is as opposed to 'keys', which are preset by module type and are directly 
tied in to the relays.

so if you have a sonoff basic, you have key1, rel1, and led1, all tied together, 
with the toggle/follow/inv/etc logic available.

If you then hook up switches to the extra GPIOs, they are not tied in to the 
relay by the module, all that work needs to be done by the central system.

I don't know if there is a way to configure a user switch to the different 
modes.

switch1 and switch2 will output different messages when they are pressed, so you 
will be able to tell them apart.

David Lang

mihalski, 2017-03-01:
The problem is that switch 1 and switch2 don't output seperate messages. It appears that only switch2 works (GPIO14) and presents as POWER via MQTT.

I am using a Wemos module so there are no relays.

mihalski, 2017-03-02:
So is there no way to use multiple switches? I appreciate that they're SUPPOSED to output different messages but they don't :/ At least not in my experience.

closed 2017-03-12 by arendst