arendst/Tasmota #127: Sonoff TH10 - Http Requests Temperature and Humidity

FabianHM, 2017-03-01:
I am looking for Http Requests by using the Sonoff TH10.
I wanted to read and monitore Temperature and Humidity Monitoring from this device.

For example like this:
- http://Sonoff-IP/cm?cmnd=status
- http://Sonoff-IP/cm?cmnd=power%20on
- http://Sonoff-IP/cm?cmnd=power%20on

Thanks
Fabian

arendst, 2017-03-01:
http://Sonoff-IP/cm?cmnd=status%2010 does the job for me:
```
STATUS10 = {"StatusSNS":{"Time":"2017-03-01T13:01:54", "DS18B20":{"Temperature":21.8}}}
```
I noticed that if syslog is enabled it might trigger a panic as it uses UDP and that might hang when a http request is made. As a workaround disable syslog with command ``syslog 0``.

I'll solve it in next release.

FabianHM, 2017-03-01:
Thanks for your very quick feedback,
Best regards
Fabian

Rudoxxxx, 2017-03-04:
After this command asking user and pasword.
Where I need to write password.

Need user=<username>&password=<password>

Rudoxxxx, 2017-03-04:
How can I need write command?

http://192.168.2.100/cm?cmnd&status%2010&user=<xxxx>&password=<xxxx>
Not working

Rudoxxxx, 2017-03-04:
http://192.168.2.100/cm?cmnd&status%2010&user=<xyaxxx>&password=<xxxx>

Rudoxxxx, 2017-03-04:
<>

Rudoxxxx, 2017-03-04:
192.168.2.100/cm?cmnd&status%2010&user=<username>&password=<password>
This not working. Can some ideas?

Rudoxxxx, 2017-03-04:
<> This symbol after user and passwors not writing in this page

davidelang, 2017-03-04:
try

http://username:password@192.168.2.100/cm?cmnd&status%2010

Rudoxxxx, 2017-03-04:
The some in my gogle chrome browser 
Need user=`<`username`>`&password=`<`password`>`"

Rudoxxxx, 2017-03-04:
May by you not using button at top right side `<>`, and autodelete text inside in `<>`?

arendst, 2017-03-04:
Maybe, perhaps, the wiki - commands page might give some information how to construct your user and password information.

BTW it only asks for a password if you configured a Web Admin Password so you should know how to proceed...

closed 2017-03-08 by arendst