arendst/Tasmota #128: Possible Instability in sensor readings?

SrKlein86, 2017-03-01:
Hi again friends.
Yesterday I updated my sonoff TH with AM 2301 sensor from 3.9.11a to 3.9.21 20170224, and since then it's been making a lot of bad readings. Here's the thing:
[image]
As you can see, the graph shows readings from a couple of days ago (feb 14th more exactly) and my sonoff is configured to do 1 reading every hour (3600s I believe is the maximum available for telemetry readings). Before updating there were some bad readings, like 1 in 3 or 4 days(which is pretty acceptable for me), but in the last day there's been 7 bad readings.
This is happening  since I updated my firmware, so that makes me believe it might be some issue with newer versions.
Hope it helps

arendst, 2017-03-01:
I've looked into it but can't explain your situation. What happens is:
- A second before telemetry time it will setup to read the sensor
- The next second it starts probing the interface for some bytes, calculates a checksum and compares it to the last received byte.
- If the checksum fails it returns the temp/hum of the previous measurement. I've seen checksum failures regularly with this kind of sensors.
- If it succeeds it will return the current temp/hum
As you do this sequence once an hour and I've seen checksum errors regularly you would receive temp/hum values from the previous measurement. I can't see this in your graph; it looks that it receives correct values all the time.
In that case the sensor returns data with correct checksum but wrong temp/hum as can be seen on the last day of your graph. I have no control over valid data if the checksum is correct...

You can check for checksum errors by enabling logging 3 like ``weblog 3`` or ``seriallog 3``. See my logging below
```
14:26:46 DHT: Timeout waiting for start signal low pulse
14:26:46 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:26:46", "AM2301":{"Temperature":22.7, "Humidity":42.1}}
14:27:47 DHT: Received 01, A4, 00, E3, 88 =? 88
14:27:47 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:27:47", "AM2301":{"Temperature":22.7, "Humidity":42.0}}
14:28:47 DHT: Received 01, A3, 00, E3, 87 =? 87
14:28:47 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:28:47", "AM2301":{"Temperature":22.7, "Humidity":41.9}}
14:29:48 DHT: Received 01, A3, 00, E4, 88 =? 88
14:29:48 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:29:48", "AM2301":{"Temperature":22.8, "Humidity":41.9}}
14:30:48 DHT: Received 01, A2, 00, E4, 87 =? 87
14:30:48 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:30:48", "AM2301":{"Temperature":22.8, "Humidity":41.8}}
14:31:49 DHT: Received 01, A0, 00, E4, 85 =? 85
14:31:49 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:31:49", "AM2301":{"Temperature":22.8, "Humidity":41.6}}
14:32:50 DHT: Received 01, A1, 00, E5, 87 =? 87
14:32:50 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:32:50", "AM2301":{"Temperature":22.9, "Humidity":41.7}}
14:33:50 DHT: Received 01, A0, 00, E5, 86 =? 86
14:33:50 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:33:50", "AM2301":{"Temperature":22.9, "Humidity":41.6}}
14:34:51 DHT: Timeout waiting for start signal low pulse
14:34:51 MQTT: tele/sonoff5/SENSOR = {"Time":"2017-03-01T14:34:51", "AM2301":{"Temperature":22.9, "Humidity":41.6}}
```

SrKlein86, 2017-03-02:
> In that case the sensor returns data with correct checksum but wrong temp/hum as can be seen on the last day of your graph. I have no control over valid data if the checksum is correct...

Yeah, that's what I thought... I just wanted to point it out since it happened right after upgrading the firmware of the device (by serial port, not OTA). 
Yesterday I updated it again to 3.9.22, just to see if anything changed, and in last 24h there have only been 2 "bad readings".
As a colleague pointed out, I was feeding the Sonoff from USB (for testing purposes), so maybe there was not enough current and that makes the sensor unstable (maybe something changed in the software and makes it consume a little bit more?).
Now I will try connecting it to AC and see if I get more accurate readings.
Will update tomorrow with any new info I get. 
Thanks for the support!

SrKlein86, 2017-03-03:
OK, it turns out that there hasn't been any "bad readings" in the last 24h, so I guess it was all about energy consumption.

SrKlein86, 2017-03-06:
Well, I'm closing this thread now, since it looks like the issue was related to power consumption.
Latest version consumes more power than the old one I had, and that was the reason that made my sensor work incorrectly.
[image]
As you can see now, there's absolutely no "bad readings" since I connected the Sonoff to AC.
Thanks again for your support.

closed 2017-03-06 by SrKlein86