arendst/Tasmota #129: settings, static IP

eriktwotoe, 2017-03-01:
Would be nice, to be able to set a static IP address, getting headache figuring it out how to do it in the code.

arendst, 2017-03-01:
Only DHCP is supported ;-)

MacWyznawca, 2017-03-01:
I use bind IP to MAC in My router DrayTek Vigor or NAS QNAP (with DHCP Server).

eriktwotoe, 2017-03-01:
OK, thanks for all the great work anyway :)

if some lost soul would like static IP. 
edit: sonoff.ino insert at about row 141
add:
// WiFi Settings
const uint8_t ipadd[4] = {192,168,x,x};  //static ip 
const uint8_t ipgat[4] = {192,168,x,x};    //local router gateway ip
const uint8_t ipsub[4] = {255,255,xx,xx}; //subnetmask

then edit support.ino add line at row about 154
WiFi.config(ipadd, ipgat, ipsub); // set static IP

// it works great for me..

arendst, 2017-03-03:
Thnx for the idea. Will be implemented in next version

closed 2017-03-05 by arendst

moonfarms, 2017-08-10:
Try tghis google search
static ip is the bane of all networks

davidelang, 2017-08-11:
static IPs can be extremely useful, if you can't keep track of what's on your 
network, it's not going to be a very good network in any case.

If I don't go with static IP addresses in devices, I go with DHCP static 
allocations.

moonfarms, 2017-08-11:
@davidelang : "  I go with DHCP static
allocations."
That's the right way to do it

davidelang, 2017-08-11:
On Fri, 11 Aug 2017, Paul Meier wrote:

> @davidelang : "  I go with DHCP static allocations."
> That's the right way to do it

There is no "one right way" in tech, any rule you try to create will end up 
having someone, somewhere run into a situation where the best thing to do is to 
break the rule.

going with static allocations in DHCP rather than static IPs on the devices adds 
a dependency to the network that can take the network down if it's not working. 
Everything is a trade-off.