arendst/Tasmota #130: Button header on Dual

hamwong, 2017-03-01:
Hi,

Thanks for this great work, I understand this question have been ask many times, just want to clarify one more time, 
I do this on Dual with original firmware,(GND/Button0/Button1/GND) 
when GND connect Button0, Relay 1 toggle, than disconnect and conect again, Relay 1 toggle,
When GND connect Button1, Relay 2 toggle, than disconnect and conect again, Relay 2 toggle,

Double click on Button0/1 won't be other action just quick on/off to mapped relay, also it won't affect another reply, Button0 only change Relay1 status and same as other,

can your code run as same way?
also can it more than toggle? means like sonoff basic with switch
GND connect Button0 > Relay 1 on
GND disconnect Button0 > Realy 1 off

Appreciate what you do on this.
Thank you and regards
Ham

closed 2017-03-05 by hamwong

jiafengwang, 2017-05-08:
I am meeting the same problem.

davidelang, 2017-05-08:
On Wed, 1 Mar 2017, hamwong wrote:

> Thanks for this great work, I understand this question have been ask many times, just want to clarify one more time,
> I do this on Dual with original firmware,(GND/Button0/Button1/GND)
> when GND [...]

the sonoff dual only has one button. If you connect switches to the gpio pins, 
then you cannot setup button0, (as it's already defined), you can setup switch0 
and switch1 though.

> can your code run as same way?
> also can it more than toggle? means like sonoff basic with switch
> GND connect Button0 > Relay 1 on
> GND disconnect Button0 > Realy 1 off

you have to use a switch and switchmode for this functionality, not a button.

David Lang

chietanjain, 2017-11-13:
The Sonoff Dual has 4 header pins to which two individual buttons (or switches) can be connected to individually control each relay. This functionality is disabled when using tasmota. Is there any way this functionality can be restored to how the stock firmware behaves?
If not is there any other 8266 based board that can do the same?
Would appreciate any help.

davidelang, 2017-11-13:
someone would have to figure out the internal programming on the chip that is 
connected to those pins and then write code to run on the ESP8266 to send the 
appropriate messages to that chip.

It's a lot of reverse engineering work. Not easy