arendst/Tasmota #132: Sonoff RF - code learnt but not triggering relay on/off

oscarliuyc, 2017-03-01:
I successfully uploaded the firmware to my Sonoff RF.  The MQTT part worked flawlessly.  It learnt an RF code with two short buttons pressed.  I think the code learning succeeded as every time I pressed the RF remote button, Sonoff RF responded by blinking the LED in red.   The problem is that the RF received couldn't trigger the change in relay state, not as what I expected.

Have I done anything  wrong?

MacWyznawca, 2017-03-01:
I have this same problem (firmware v3.9.18 -3.9.22).

arendst, 2017-03-03:
Are you using the preferred ITEAD RF transmitter? I was unable to get it working with my non-ITEAD preferred RF transmitter.

MacWyznawca, 2017-03-03:
My transmitter work after I change firmware from „China” to Tasmota. Red LED blink same like with „China” firmware.

oscarliuyc, 2017-03-05:
Hi MacWyznawca,

What was the "China" firmware?  After changing to Tasmota, did the rely state change with the red LED light ?

MacWyznawca, 2017-03-05:
„China” = original from Itead. If I press button on 433 MHz remote (works with Itead firmware) red LED light but power  doesn't switch.

oscarliuyc, 2017-03-06:
So my guess would be that the RF receiver module did pair with the remote but somehow it couldn't trigger the Tasmota firmware to change the relay state.

dobrMAN, 2017-03-15:
I have some issue on Sonoff RF, but after power reset on old firmware (version 3.9.3).
I pair my RF transmitter by pressing Sonoff button twice and then press on transmitter button. It pair ok and work great, but after power shutdown and boot on power up, RF button do not swith relay. Red led blink when i press RF button, but relay do not switch. After clear pairing (3 short button press on Sonoff) and repairing button it work again. It is no good, i must repair button any time after power down :(

davidelang, 2017-03-18:
pressing the sonoff button twice doesn't set pairing in this firmware, did you try following the instructions at https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF ?

MacWyznawca, 2017-03-19:
That does not describe how to pair, but how to upload firmware. I have not shorted permanently these contacts.

off14, 2017-05-04:
what to select in device configuration for GPIO14 in sonoff rf? can be used for wired switch together with RF? does RF module takes GPIO14??

maddhin, 2017-10-19:
I just paired my (ITEAD recommended) remote with the RF successfully by keeping the remote button pressed and pushed the RF button twice. The lights are changing several times in different colors (red, orange, green) and I think I stopped pushing the remote button when the green light was stable on. 
So, version 5.8.0 seems to work (although I needed several tries to make it work).

closed 2018-05-07 by stale[bot]