arendst/Tasmota #133: http upgrade "broken pipe" error

jesusvallejo, 2017-03-01:
it seems that the sonoff closes the link before the transfer is completed and i get this error in my http server and also the sonoff does not update . any clue why is this happening or if maybe is a bug.

thanks

ThomDietrich, 2017-03-03:
I'll try to help here... sadly your description doesn't paint a clear picture.

Which error do you get?
What does "sonoff does not update" mean?
Which openhab upgrade method are you talking about?

Please describe the problem you are seeing as detailed as possible. You may also upload screenshots.

@arendst feel free to tag me on openHAB related issues.

jesusvallejo, 2017-03-07:
@ThomDietrich i can see the error in the http server log, the error code is broken pipe.and it means (according to google) that the client has brocken the link before clompleting the transference of the file.

"it does not update " means that it does not apply the firmware.bin , so if i want to upgrade from 22 to 23 it stays in 22.

the openhab upgrade method is posted in the tasmota wiki in the openhab part.

ThomDietrich, 2017-03-07:
This answer didn't help much. You need to phrase your error description clearly and give concrete evidence. 
Did you try a more direct upgrade approach? https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade
The obvious choice would be to option (1) upgrade directly through the http web interface of your sonoff-tasmota module.

jesusvallejo, 2017-03-07:
this is the error in the server, in the sonoff there is no way to say if it went ok or not.
`----------------------------------------
Exception happened during processing of request from ('192.168.1.31', 19166)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------
`
yes thats the way i ussualy do it but i saw this on the wiki and now i prefer to do it this way , and if i learn something in the procces better.
and the wifi connection is fine all of the have between 90% and 80% of signal.

ThomDietrich, 2017-03-07:
@jesusvallejo thanks for posting the error log.
I'm seeing python issues. Which server are we talking about? Which file did you take these lines out from? Are you using Mosquitto as your MQTT server and openHAB in a standard installation with the maintenance setup as described [here](https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB#maintenance-actions)?

jesusvallejo, 2017-03-07:
@ThomDietrich  yes i use mosquitto and works right , openhab is installed with apt-get method on raspbian , works right . and yes it has that setup.in fact i thin openhab and the server do well their job so in my opinion it might be sonoffs problem.the server is simple python http server `python -m SimpleHTTPServer 9090` and the log is in real time on the console.

jesusvallejo, 2017-03-07:
`00:00:31 MQTT: stat/sonoff1/RESULT = {"OtaUrl":"http://192.168.1.69:9090/firmware.bin"}
00:00:31 MQTT: stat/sonoff1/RESULT = {"Upgrade":"Version 4.0.1 from http://192.168.1.69:9090/firmware.bin"}
00:00:32 HTTP: Webserver stopped
00:00:32 HTTP: Webserver active on sonoff2-0317.local with IP address 192.168.1.31
00:00:34 HTTP: Webserver stopped
00:00:34 HTTP: Webserver active on sonoff2-0317.local with IP address 192.168.1.31`

this is sonoffs console during the process

ThomDietrich, 2017-03-07:
aaaah we are talking about the local http server hosting the firmware file. That's the piece of info missing here ;)
Okay now we are talking. This could indeed be related to the way Sonoff is retrieving the file. Did you try @smadds URL? If I find the time, I can try to reproduce this this evening at home

jesusvallejo, 2017-03-07:
@ThomDietrich yes but when i tryed it , it was not operative

ThomDietrich, 2017-03-07:
Maybe it's easiest to use another http server. If you are already hosting an openHAB instance, move the file to `/etc/openhab2/html/firmware.bin`:

> Serve your own static html pages or resources from here.
> Files stored in this folder will be available through the HTTP server of openHAB, e.g. "http://device-address:8080/static/image.png".
> Resources for sitemap elements (image, video,...) can also be provided though this folder.

jesusvallejo, 2017-03-07:
@ThomDietrich now it works like a charm , any idea why?? and also do you now if there is any way to get the last firmware from github and copy it into html directory in an automatic way.

thanks a lot

ThomDietrich, 2017-03-07:
https://github.com/arendst/Sonoff-Tasmota/issues/19#issuecomment-277537341
Choose one link, then use cron and wget to download the file every e.g. evening.

If your issue is resolved, please close it. Glad it's now working for you ;)

jesusvallejo, 2017-03-07:
yeah thanks

closed 2017-03-07 by jesusvallejo