arendst/Tasmota #135: Sonoff Basic GPIO14 not toggling

gid204, 2017-03-02:
I'm having an issue with my SONOFF basic. 

Switch mode in user_config.h is set to toggle. I have it set to the sonoff basic module and gpio14 set to switch 1 in the web console configuration.

However, when I ground gpio 14 it acts as a pushbutton. i.e it'll change state for as long as I hold it but as soon as I let go it switches back.

Any help is greatly appreciated!

hamwong, 2017-03-03:
use switchmode command

gid204, 2017-03-04:
Hey thanks for your reply!
Sorry if this is a stupid question but what do you mean use the switch mode command?
I've got switch mode set to toggle in user_config.h ....

thanks!

hamwong, 2017-03-04:
You may goto web-ui and type switchtopic command
Further detail may look here/wiki/commands

hamwong, 2017-03-04:
I am not very good in English as its not my mother language, I don't know do you understand what I said or not?

gid204, 2017-03-04:
I think I do understand, I will try again. Thank you!

gid204, 2017-03-04:
I did as you said but it still does not work :(
I went to web UI > console and use switchmode<0>
but it still does not work :(
any other suggestions?

hamwong, 2017-03-05:
no.... you miss something,
<0> means switch no. as you can max assign 4 switch,
you should first just type swichmode and see what it come out, 
it will tell you your current switch mode
than you should try with difference no. see which one is your want

hamwong, 2017-03-05:
So You should do it as switchmode first, than it will tell you that your current mode like switchmode 0
Than you may try switchmode 1 to 4

closed 2017-03-12 by arendst