arendst/Tasmota #137: Access to analogue pin

jaysuk, 2017-03-02:
On the wemos d1 mini etc, as well as digital pins, there is 1 analogue pin.
Would it be possible to be able to configure this as report the value back?
This value would be between 0 and 1023.

khcnz, 2017-03-12:
I was looking into this - the adc is also available on the sonoff dev and accessing the value from firmware is trivial but integrating it into everything else a little harder

How do you propose we would report the data? Using the tele command with the same tele-period? How would you enable/disable the value - perhaps an enable/disable flag on the module selection/gpio configuration page? The ADC can also be configured to report the input voltage to the esp  -so perhaps a drop down of ADC Enabled field with the following options "Off", "A0", "VCC"

arendst, 2017-03-13:
@khcnz now that is seriously usefull. How does anyone want to receive the information?

davidelang, 2017-03-13:
I would start off by reporting the raw value (0-1023)

On Mon, 13 Mar 2017, Theo Arends wrote:

> @khcnz now that is seriously usefull. How does anyone want to receive the information?
>
>

khcnz, 2017-03-13:
As A0 repor will be 0-1024 from memory. VCC reports the voltage (so approx 3300). I am still thinking something in the tele command will be most useful - I will have a PR in the next couple of hours to review.

khcnz, 2017-03-13:
Well I got this mostly implemented so it can be configured on the GPIO page.... only to (rediscover) that you can't configure the VCC/A0 dynamically; You can only set ADC_MODE(ADC_VCC) outside of a function. So we are left with a preprocessor to select between VCC and A0 - which makes the point of dynamically selecting it within the module configuration GPIO PINS page a little pointless...

It also looks like we would likely need a correction factor https://github.com/esp8266/Arduino/issues/721

khcnz, 2017-03-14:
More useful info http://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations

I propose

1. PWM1-4 user programmable (so the GPIO pins can be configured to out a PWM signal via analogwrite) 
2. A0 to be configurable to on/off so it can be read with analogRead. When configured on will return result on status 10 command.
3. Allow a new command PWM1..4 that can set duty with PWM1..4 0..1024
4. A complile time #define to allow VCC to be read rather than A0 - possibly this could returned on status 1?

khcnz, 2017-03-16:
@jaysuk  I think this one can be closed now?

jaysuk, 2017-03-16:
It can. Will close it now

closed 2017-03-16 by jaysuk

fttx, 2017-08-04:
Hi,

is it possible to publish the analog value of pin A0 as telemetry data? Something like this: tele/sonoff/A0 -> 0.99
If yes is it possible to redefine the range of the value? Like [map](https://www.arduino.cc/en/Reference/Map)?

davidelang, 2017-08-04:
the A0 pin reading has been reported since version 4.something. You can't scale 
it in the firmware, do that in the software receiving it.

fttx, 2017-08-04:
Ok, thanks, but what is the topic where it gets published?

davidelang, 2017-08-04:
it would be part of the tele/ reports. You also will see it as a result of a 
status 0 command

roesl, 2017-12-21:
How I can activate the Analog pin for Wemos D1 mini pro ? - I dont find the pin in configuration menu

K1WIZ, 2019-01-14:
Hi, I enabled the ADC pin on compile, and I can see the raw value now displayed on the main web UI.   I want t o send this value as a "custom sensor" value to Domoticz.   I have a Custom Sensor configured on Domoticz ready to receive the data, and I would think I would find a way to establish the IDX destination for the A0 pin value to be sent to that Domoticz device - but I can't find any selection for the A0 pin on the Module Parameters page on Module Config.    It would be really useful to be able to send a raw analog value to Domoticz.   My use case is monitoring a water level that uses a simple conduction probe connected to A0.   I could then use this raw data to create actionable results via Domoticz.   How can I achieve this in Tasmota?

arendst, 2019-01-14:
Sending every change to MQTT and/or domoticz would flush bothservers.

There is a feature I use to control the dimmer of led by using the ADC input of another device. To use this you will have to define a rule. This rule can also be used not to send the ADC info to the led  but to Domoticz.

This is my led rule:
```
on analog#a0div10 do publish cmnd/ailight/dimmer %value% endon
```
As you can see there is a trigger that activates with every tenth change in the analog input.

Using the rule examples from the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#examples) you must be easily able to send these changes to domoticz with something like:
```
on analog#a0div10 do publish domoticz/in {"idx":134,"svalue":"%value%"} endon
``` 
You'll have to find the correct domoticz sensor svalue syntax depending on the type of custom sensor you defined. The idx is the idx of the sensor in Domoticz.

K1WIZ, 2019-01-14:
Theo,  Thank you for your assistance.  Your suggestion helped me to successfully report the values to a custom sensor object in Domoticz.  I can see the value sent whenever it changes on A0.   I would like to be able to send the value whenever the telemetry period fires.   Currently, in the console, I see the value returned every few minutes.   The way it is now, Domoticz only gets a new value when there is a delta.  Here's what I have in the console:

> 19:31:51 MQT: tele/sonoff/STATE = {"Time":"2019-01-14T19:31:51","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"MIND CONTROL BG","BSSId":"0E:EC:DA:38:D5:8A","Channel":1,"RSSI":98}}
19:31:51 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-14T19:31:51","ANALOG":{"A0":1024}}
19:36:51 MQT: tele/sonoff/STATE = {"Time":"2019-01-14T19:36:51","Uptime":"0T00:05:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"MIND CONTROL BG","BSSId":"0E:EC:DA:38:D5:8A","Channel":1,"RSSI":98}}
19:36:51 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-14T19:36:51","ANALOG":{"A0":1024}}

> 

Here's what my rule1 looks like:

> 19:38:42 CMD: rule1
19:38:42 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":423,"Rules":"on analog#a0div10 do publish domoticz/in {"idx":421,"nvalue":0,"svalue":"%value%"} endon"}

> 

Can you suggest how I send this to Domoticz when the telemetry runs?   Thanks a lot and thank you for the Tasmota project!   I love it.

arendst, 2019-01-14:
Hint: on tele-analog#a0 do publish ...