arendst/Tasmota #138: What happens when a switch is turned on and off

amandus99, 2017-03-02:
Hi,
my plan is to use a sonoff with Tasmota for my kitchen lamp. What will happen if I will turn on the lights with a common switch? Would it be possible that the sonoff is set to on when it is powered on by the common switch like a Philips Hue?

davidelang, 2017-03-02:
The basic sonoff has no idea if any power is being drawn through it, so it can't 
tell if the lamp is turned on at the lamp or not. The sonoff pow can measure if 
power is going through it, but if it's off, there will be no power going through 
it and it won't be able to tell if a switch at the lamp is turned on or not.

If you turn it off by a wall switch, then the sonoff looses power as well. When 
it gets power, you can set if it will get turned on or not.

MacWyznawca, 2017-03-03:
If You connect external switch to GPIO14 and correct setup that switch, Sonoff know switch state, and send MQTT comand.
https://www.instagram.com/p/BQNP_l5DtuP/?taken-by=macwyznawca

closed 2017-03-12 by arendst