arendst/Tasmota #140: some basic question about WS2812

hamwong, 2017-03-03:
HI, 

I just bought a WS2812 LED stripe as you add this feature, I got some question I think that should be simple, 

1) my LED stripe should have around 300 pixels, I found the pixels in config max is 255, so once I set dimmer 0, first 255 turn off but rest still out of control, can it be more than 255? 

2) I found if I set dimmer more than 30 or 50 ( I don't quite remember), I set color white, first few pixels is white than pixel fading to orange , is that due to not enough power from power converter? but it's 5V 5A,

3) in user_config.h there is WS2812_DMA, and youtube got few demo of DMA effect, looks cool, question is how to use DMA feature?

altelch, 2017-03-03:
1) Set WS2812_MAX_LEDS in sonoff.ino to the required value. The variable holding the pixel count is uin16_t so it's large enough for 65535 LEDs. That should fix your problem. Guess you have 5m of 60LED/m strips. 

2) You have 300LEDs @max 60mA/LED -> 18A so 5A is much to weak.

3) DMA stands for Direct Memory Access and has nothing to do with effects but the way the ESP8266 controlls the LED strip.

hamwong, 2017-03-03:
Thanks for your detail answer, just 1 more quick question, 
as description said use DMA will cost extra 1K mem, so if activate DMA will it be any advantage on sonoff controller? I mean as memory usage is very limited on sonoff, is it worth to activate DMA and what can I got after activate?

altelch, 2017-03-03:
In DMA mode the ESP8266 does the serial protocol with the LEDs in Hardware and in the background instead doing software bit banging. This might help if you run into timing problems on longer strips as updating every LED takes some time. Never ran into that problem so I can't tell you at which point switching to DMA might be helpful.

hamwong, 2017-03-03:
XD, Thanks for your answer, I got difficult on choices, I'll go for a try on DMA see what will be difference.
Best regards

hamwong, 2017-03-03:
Hi @arendst 

Recently I just bought some wifi led controller, those device use esp8266, are you interest on support it on you firmware, I may send few devices to you if you are interested, you may send email to me hamwong338@gmail.com
[url=https://postimg.org/image/q801yy37x/][img]https://s22.postimg.org/q801yy37x/IMG_4542.jpg[/img][/url]

[url=https://postimg.org/image/9yeh9shxp/][img]https://s22.postimg.org/9yeh9shxp/IMG_4543.jpg[/img][/url]

[url=https://postimg.org/image/5qjp11gi5/][img]https://s22.postimg.org/5qjp11gi5/IMG_4544.jpg[/img][/url]

[url=https://postimg.org/image/7wdzvjjyl/][img]https://s22.postimg.org/7wdzvjjyl/IMG_4545.jpg[/img][/url]

hamwong, 2017-03-03:
Also I bought some sonoff touch like wall switch, it got from 1way to 4 way, got LN version and L only version, it use itead software for control, if u interest, I can also sent to u

[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>

hamwong, 2017-03-03:
Hi, @altech 
I found maybe pixels limit is not the main reason, I change the pixels to 300 and found led status no chagne, than I discover on LED #247, it's dead, if there is dead led, will it makes sonoff not able to order/control led after dead led?

joshuagti, 2017-03-03:
@hamwong , where did you get them light switches from? cannot find them anywhere

davidelang, 2017-03-03:
given that there is active electronics in each LED that set the brightness of 
itself, and then one clock later pass the data to the next LED, it is not 
surprising that a dead LED would break everything after it in the chain.

hamwong, 2017-03-03:
from China, Taobao chinse interface

hamwong, 2017-03-03:
price is not expensive, only controller I think around 6USD

hamwong, 2017-03-03:
@davidelang, as there is a cutting place for each LED, if I cut out the damage one and connect the well condition part, will it works?

davidelang, 2017-03-03:
As I understand things, it should.

David Lang

On Fri, 3 Mar 2017, hamwong wrote:

> @davidelang, as there is a cutting place for each LED, if I cut out the damage one and connect the well condition part, will it works?

hamwong, 2017-03-04:
HI, @davidelang 
Thanks, will retry later.

joshuagti, 2017-03-04:
@Hamwong could you send me a link? Can't seem to find them

hamwong, 2017-03-05:
@joshuagti you look for wall switch?
do you read/write chinese? I think the site is a bit hard for you to buy, it's for chinese only(I think)

anyway, you may try on below link, currency is in China dollar and Good Luck
https://item.taobao.com/item.htm?spm=a312a.7700846.0.0.UIWvdY&id=537843786441&_u=51904rq0402

hamwong, 2017-03-05:
@joshuagti if you really have bought it later and get it hacked, please kindly let me know how to, you may directly send me an email hamwong338@gmail.com, without much hardware knowledge and skill, I have broke one when attach wire directly to esp8255 -.-.  As @arendst looks not interest all, I am closing this case

closed 2017-03-05 by hamwong