arendst/Tasmota #141: Device retaining settings from previous flashes?

radiohe4d, 2017-03-03:
I had a few devices flashed with the older software, which i setup things like name and wifi within the user_config.h file. Now that I've flashed the new software I've found that some of those old values for things like wifi have persisted after the flash.

In the wifi config page the header would contain something like "AP1 SSId (TALKTALK57F)" with the TALKTALK wifi being the old wifi from the previous flash.

So my question is, is there a way to clear the Sonoff completely or is this normal behaviour; If so, why?

Cheers.

arendst, 2017-03-03:
Retaining settings is a major feature of this software! It allows for seamless installations of new versions while still introducing new features.

If you want to reset the settings just change define CFG_HOLDER in user_config.h and all your precious settings will be rest to defaults. I see no reason why...

davidelang, 2017-03-03:
retaining things when doing an OTA flash is a great feature, but if you manage 
to do something wrong in the settings and have to go back to flash via serial, 
you really do want to erase anything that's there rather than use potentially 
corrupted configs.

or if you are flashing from some other non-standard firmware, you don't know 
what is in those areas.

David Lang

arendst, 2017-03-03:
Before using the settings the software checks for valid data (comparing the CFG_HOLDER value in flash with the one in the code) and if it miss compares it writes default data to flash.

radiohe4d, 2017-03-03:
Sorry, i fully understand that retaining settings between updates is essential and i would not change this. 

But in my case i want a fresh install, whilst updating using FTDI over the old software and removal of any old settings. This TALKTALK wifi no longer exists, so i dont want my devices to retain these settings at all for this clean install.

Obviously after i flash the new software i want these settings to be retained even after OTA updates.

SrKlein86, 2017-03-07:
I've gone through this too (been trying with different SSIDs and MQTT topics), and even with a message on the header of userconfig.h, it's kinda hard to notice this small detail. Thank God there's a lot of good people here that will give us support^^
As @davidelang and @arendst stated, you just need to modify the CFG_HOLDER variable and you are good to go!

arendst, 2017-03-07:
Glad we could help. I do mention the situation at the top of ``user_config.h`` but hey, who reads now a days?
```
/*********************************************************************************************\
 * User specific configuration parameters
 *
 * ATTENTION: Changes to most PARAMETER defines will only override flash settings if you change
 *            define CFG_HOLDER.
 *            Most parameters can be changed online using commands via MQTT, WebConsole or serial
 *
 * Corresponding MQTT/Serial/Console commands in [brackets]
\*********************************************************************************************/
```

closed 2017-03-08 by arendst