arendst/Tasmota #142: Can't change wifi hostname from %s-%04d

chris-thorn, 2017-03-03:
First let me say how fantastic this project is - I really appreciate your hard work!

I have several Sonoff basic modules around the house, and for each one the wifi hostname is set to `%s-%04d`, which generates a name such as `living_room_led_tree-3602`, but I cannot for the life of me work out how the four digits following the dash are generated!

For example, the MAC address of the Sonoff in the above example is `5C:CF:7F:95:6E:12`, so where does it get the 3602 from?

I've tried changing the wifi hostname in the Configuration > Configure WiFi > Hostname textbox from `%s-%04d` to `%s-%06X`, because I would prefer to link the wifi hostname to the MAC address of the device for easier identification, but after saving the configuration and restarting the Sonoff the original hostname `%s-%04s` persists.

It's not a huge issue, but I just wonder whether I'm doing something wrong.

[image]

Thanks again!

Chris

arendst, 2017-03-03:
Get you programmer calculator and enter the decimal value of 3602. Then select hex notation and you will be surprised to find that it look the same like the last three digits of your mac address E:12

chris-thorn, 2017-03-03:
Aaaaah I tried the last four digits in hex notation, but didn't try the last three!

I just tried changing the name by manually typing in `sonoff-basic-956e12` and the name has stuck - it works perfectly.

Thanks so much!

Chris

closed 2017-03-03 by chris-thorn

tlc76, 2018-05-15:
Hi Tom,

I would like to control the naming convention solely via the PROJECT variable before writing your (by the way, GREAT!) firmware to each of the Sonoff devices I (will) own.
Would there be any issue if I change the WIFI_HOSTNAME definition (file sonoff.h) from "%s-%04d" to "%s"?

Many thanks,
Cristian

micw, 2020-02-02:
@arendst Just to be clear, changes to the wifi hostname template are not persistent and it's by intent? In this case it would make sense to set this input field read-only.

@tlc76 My workaround is to set "topic" to "tasmota" only and append the XXXX in "full topic".

mateuszdrab, 2020-03-07:
Hey guys, I'm also looking for a way of templating the host name because I have a large deployment of Sonoffs (around 40, at home and each time I provision one, I had to manually configure the topic and hostname). The topic is now overridden in my_user_config but it's a shame there's no field for hostname. More annoyingly, when I change the WIFI_HOSTNAME[] variable to "%06X" I get a decimal representation of the MAC, not the hex which works fine in the topic. Isn't this a bug?

mf-1, 2020-11-10:
Is there a way to set hostname to `%s` only? I want to remove the `-%04d` their. Is it only possible by self-compiling as @tlc76 has done? Or is there something planned at the main project?

Background is to can point the browser to this name without knowing the 4 digits.

Uriziel01, 2021-01-17:
I'm facing the same issue like @mf-1 setting the hostname to `%s` would allow me for a robust implementation where when a device fails (and they will fail, I'm having over 50 of them) all the checks and alerts in router stays intact.

osousa, 2021-02-24:
> I'm facing the same issue like @mf-1 setting the hostname to `%s` would allow me for a robust implementation where when a device fails (and they will fail, I'm having over 50 of them) all the checks and alerts in router stays intact.

You need to remove the **if** condition on xdrv_01_webserver.ino 

lines [1792-1794] : 

```
void WifiSaveSettings(void)
{
  char tmp[TOPSZ];  // Max length is currently 150

  WebGetArg(PSTR("h"), tmp, sizeof(tmp));
  SettingsUpdateText(SET_HOSTNAME, (!strlen(tmp)) ? WIFI_HOSTNAME : tmp);
  if (strchr(SettingsText(SET_HOSTNAME), '%') != nullptr) {
    SettingsUpdateText(SET_HOSTNAME, WIFI_HOSTNAME);
  }
  . . .
  . . .
```

jeroenst, 2021-07-17:
> Is there a way to set hostname to `%s` only? I want to remove the `-%04d` their. Is it only possible by self-compiling as @tlc76 has done? Or is there something planned at the main project?
> 
> Background is to can point the browser to this name without knowing the 4 digits.

I also can't remove the -%04d from hostname. As soon as I put in %s the -%04d is always added after saving :-(

This issue occurs in Tasmota 9.4.0.4

uxhamby, 2021-11-05:
Hi Theo,

Great firmware pkg btw. Thanks!

I think I am not understanding something here however.

I have a WeMos D1 mini Pro v1 device running:

Program Version | 9.5.0(tasmota)
Build date 2021-06-17T08:26:35
Core SDK 2_7_4_9/2.2.2-dev(38a443e)

It's MAC address is: 84:F3:EB:B3:F3:E4

It gets default hostname: tasmota_B3F3E4-5092

So, from the above discussion:

%s works out to be B3F3E4, no problem

but %04d works out to be: 5092

but 5092 is Hex 13e4 not F3E4.

I see that F3E4 is bigger than 4 digit decimal so is not a good candidate but I am confused as to how the algorithm works for calculating %04d in such a case. I see that max 4 decimal digits are called for but the apparently arbitrary selection of 13e4 unsettles me about relying on my understanding for future calculations as I note that hex 23e4 also works out to be a 4 digit decimal at 9188.

Is there a standards document you could refer me to on this?
 
Thanks & best regards,

Brian H.

barbudor, 2021-11-05:
%04d only gets 13 bits (mask 0x1FFF) from the MAC address

and you are not totally exact on the 1st one : %d get the topic which by default is initialized at "tasmota_%06X" so takes the  last 6 Hex digits from the MAC.

Hope this helps

uxhamby, 2021-11-06:
Yes, thanks.

mateuszdrab, 2021-11-11:
> You need to remove the **if** condition on xdrv_01_webserver.ino
>
> lines [1792-1794] :
>
> ```
> void WifiSaveSettings(void)
> {
>   char tmp[TOPSZ];  // Max length is currently 150
>
>   WebGetArg(PSTR("h"), [...]

This is pretty much what I was doing when I was self-compiling my firmware but I decided to not compile and more and I'm also struggling because no matter what I put in the hostname field, if it contains % it gets reverted to the default. Is this a bug?

netstx, 2022-10-25:
Does anyone know if there has been updates to this in version 11 or 12?

mateuszdrab, 2022-10-25:
> Does anyone know if there has been updates to this in version 11 or 12?
> 
> 
> 
> EDIT: sorry, didn't realize issue was closed

I gave up and adopted the default hostname pattern...

jeroenst, 2022-11-06:
> You need to remove the **if** condition on xdrv_01_webserver.ino
>
> lines [1792-1794] :
>
> ```
> void WifiSaveSettings(void)
> {
>   char tmp[TOPSZ];  // Max length is currently 150
>
>   WebGetArg(PSTR("h"), [...]

Can/may this be changed in master? I can't use the hostname variable..

10:44:33.459 CMD: hostname TASMOTA-%s
10:44:33.472 MQT: home/TASMOTA_PLUG004;/stat/RESULT = {"Hostname":"%s-%04d"}

10:45:26.131 CMD: hostname %s
10:45:26.147 MQT: home/TASMOTA_PLUG004;/stat/RESULT = {"Hostname":"%s-%04d"}

My hostname setting is ignored by Tasmota

I fully agree that the default setting is %s-%04d but when changing it, everything should be allowed.

mateuszdrab, 2022-11-06:
> Can/may this be changed in master? I can't use the hostname variable..
>
>
>
> 10:44:33.459 CMD: hostname TASMOTA-%s
>
> 10:44:33.472 MQT: home/TASMOTA_PLUG004;/stat/RESULT = {"Hostname":"%s-%04d"} [...]

This is how I used to work around the issue when I was self compiling Tasmota but now I've moved to using stock builds.

Ideally a PR to master to remove the if condition would be the solution, question is, why was it put there in the first place.

barbudor, 2022-11-06:
You can set hostname to any string, just not one with a %
```
00:00:05.007 HTP: Web server active on the-most-stupid-hostname with IP address 192.168.168.105
```