arendst/Tasmota #143: Version 4.0.0

KPK75, 2017-03-03:
After upgrade from 3.9.22 to 4.0.0 MQTT is not working. Sonoff basic should get switch state from MQTT but it's unable to maintain connection. The only difference I see in mosquitto.log is this third line (bold)
Ver 4:
`1488545465:     cmnd/DVES_954FB9/# (QoS 0)`
`1488545465: Sending SUBACK to DVES_954FB9`
**`1488545541: Sending CONNACK to DVES_954FB9 (0, 0)`**
Ver 3
`1488545541:     cmnd/DVES_954FB9/# (QoS 0)`
`1488545541: Sending SUBACK to DVES_954FB9`
`1488545541: Received PUBLISH from DVES_954FB9 (d0, q0, r0, m0, 'stat/sonoff3/RES`

Any ideas ? After downgrade everything works fine.

jaysuk, 2017-03-03:
> I was unable to open the web settings after this update and the device
> would not connect to the MQTT server either. Had to downgrade to fix it.

MQTT and the web pages work fine on all my sonoffs after upgrading to 4.0.0
from 3.9.22.

arendst, 2017-03-03:
Mine too...

radiohe4d, 2017-03-03:
Would editing any of the default values in the user_config.h prior to upload have any adverse affects?

arendst, 2017-03-03:
No, it shouldn't but what value did you change?

radiohe4d, 2017-03-03:
Both WiFi values, MQTT host/password, Web Password, Emulation type (Wemo) and disabled mDNS Discovery and all the sensors.

arendst, 2017-03-03:
Right. Why would you want to change all these values to upgrade from 3.9.22 to 4.0.0? There is no parameter change between these two versions. So the correct way is to upgrade first to 4.0.0 and then start fiddling around with your settings although I can't understand why people want to do this as most of these things can be done online.

jaysuk, 2017-03-03:
I always update the user_config with the right details and then compile a
couple of different builds. Purely for if I have any issues and have to
reset a device there's less for me to setup. So no, changing values in
user_config shouldn't have effected the upgrade

radiohe4d, 2017-03-03:
Because i have 10+ devices in front of me that i am either flashing from stock or upgrading from a previous version (which i didn't config OTA updates on). So to edit this file to be my default setup saves me going into every single device and adding the same parameters, such as wifi and passwords and mqtt logins. 

This way i only have to change the name of the device on the web page and all updates from now on will simply use the ota and ill never have to touch the config file.

Greg2002, 2017-03-03:
After upgrading I can't start Web interface at all. Return back to  3.9.22 -works fine. Reflash - the same story. Only hear clicks on relay module on loop.

arendst, 2017-03-03:
@Greg2002  what is the output on the serial interface during the loop?

legotheboss, 2017-03-03:
I am having the same issue as @Greg2002

legotheboss, 2017-03-03:
Just downgraded to 3.9.20 and issue has disappeared, the rst code on the serial output was 4...

arendst, 2017-03-03:
Was there no stack dump?

legotheboss, 2017-03-03:
What does the stack dump look like?

Greg2002, 2017-03-03:
rst cause:4, boot mode:(3,7)

arendst, 2017-03-03:
That is no stack dump

davidelang, 2017-03-03:
If you want config changes to be applied when you flash a device, you need to 
change this line in user_config.h

#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters

otherwise it continues to use the prior config.

you can force it to switch the the compiled-in config from whatever it has by 
holding the button down for 4 seconds.

David Lang
k

Greg2002, 2017-03-03:
rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

arendst, 2017-03-03:
It looks like hardware wdt as explained here https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.md

There will be no stack dump  so it is hard to debug.

For me to investigate I need more info that you may be able to provide using seriallog option 3 or 4.

Greg2002, 2017-03-03:
No stack, possible the problem with watchdog

Greg2002, 2017-03-03:
tested with 2 modules - the same story

arendst, 2017-03-03:
What happens if you change MESSZ to 240 on line 148 in sonoff.ino?

Greg2002, 2017-03-03:
Decoding 3 results
0x40201c14: setup at C:\Users\P6600\Documents\Arduino\!Ready projects\Temp\sketch_mar04a/sketch_mar04a.ino line 5
0x40201fe8: loop_wrapper at C:\Users\P6600\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100114: cont_norm at C:\Users\P6600\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

Greg2002, 2017-03-03:
3ffef2f0:  3fffdad0 00000000 3ffee2d4 40201c14  
3ffef300:  feefeffe feefeffe 3ffee2f0 40201fe8  
3ffef310:  feefeffe feefeffe 3ffee300 40100114

arendst, 2017-03-03:
Where does the sketch_mar04a line 5 come from? Not sonoff!

Greg2002, 2017-03-03:
Yes I tested a sample with Arduino ESP8266/ESP32 Exception Stack Trace Decoder. Found a stack error, but when I load sonoff v/4 - no stack!

Greg2002, 2017-03-03:
Let's continue tomorrow

Greg2002, 2017-03-03:
Reset reason -Power on (v3.9.22) and
Reset reason -Watchdog(v.4.0.0) !!!

Greg2002, 2017-03-04:
Hello! Changed MESSZ to 240 and now it works fine! V.4.0.0

Greg2002, 2017-03-04:
and reset reason ( in information tag) changed to Power on

Greg2002, 2017-03-04:
Tested with MESSZ  320. It's max stable number of characters in JSON message string for me

davidelang, 2017-03-04:
what is MQTT_MAX_PACKET_SIZE set to on your system?

On Fri, 3 Mar 2017, Greg2002 wrote:

> Tested with MESSZ  320. It's max stable number of characters in JSON message string for me

Greg2002, 2017-03-04:
512

Greg2002, 2017-03-04:
you can't compile before you change it to 512

davidelang, 2017-03-04:
The prior documentation aimed for a MQTT_MAX_PACKET_SIZE of 400 and a MESSZ of 
240 with a TOPZ of 60. this resulted in a buffer of 100 between MESSZ and MAX*, 
of which 60 bytes is the TCP headers. This is why the calulation on line 149 is 
doing -60 -40

on your system, the minimum you need is a difference of 132 (72 bytes not 
counting TCP overhead).

I don't know the MQTT over-the-wire-protocol to know what the right value is 
(there are some variable sized headers in there to mess with things)

we could just change the '40' to '72', but I hate to make changes like this on 
the 'it works' basis :-)

It's far better to find the correct value.

are you doing authentication or TLS for your MQTT connection?

On Fri, 3 Mar 2017, Greg2002 wrote:

> you can't compile before you change it to 512

Greg2002, 2017-03-04:
No TLS, ordinary MQTT connection

arendst, 2017-03-04:
Glad we are down to the MESSZ value. There is more to it than just a buffer size; it is used by most char defines and increasing it also changed the compile size I cannot explain just now. As memory is tight on the ESP8266 it may well be a problem with this too.

For now I will change the value back to 240 in the upcoming release as I have no need for a bigger value anyway and investigate further what causes the hardware WDT.

Greg2002, 2017-03-04:
OK. Thanks. In  my case it works without any problem up to MESSZ 320

altelch, 2017-03-04:
Hi, same here. Switched MESSZ to 320 and the Sonoff works again. Before ran into the same exceptions as shown above.

davidelang, 2017-03-04:
I'm happily running 3.22 with max packet set for 1400 abd messz od 1240 so it's 
not quite as simple as just large bufers for messages.

Greg2002, 2017-03-05:
4.01 works fine with MESSZ  360

closed 2017-03-08 by arendst