arendst/Tasmota #144: compile error

kfonda, 2017-03-03:
Hi all,
  For the last several versions, I've been getting this error when trying to compile using PlatformIO on Windows 7...
```
D:/Development/GitHub/Sonoff-Tasmota/sonoff/sonoff.ino: In function 'void stateloop()':
D:/Development/GitHub/Sonoff-Tasmota/sonoff/sonoff.ino:3529:19: error: 'sl_blank' was not declared in this scope
D:/Development/GitHub/Sonoff-Tasmota/sonoff/webserver.ino: In function 'void handleUploadDone()':
D:/Development/GitHub/Sonoff-Tasmota/sonoff/webserver.ino:1963:15: error: 'sl_blank' was not declared in this scope
D:/Development/GitHub/Sonoff-Tasmota/sonoff/webserver.ino: In function 'void handleUploadLoop()':
D:/Development/GitHub/Sonoff-Tasmota/sonoff/webserver.ino:2053:15: error: 'sl_blank' was not declared in this scope
```

I've been adding `void sl_blank(byte state);` to the top of sonoff.ino in order to get it to compile, and everything seems to work fine after that. I've gone as far as to uninstall PlatformIO and delete the Tasmota code and reinstall fresh without making any changes at all and I still get the error. 

Can anyone give me any ideas as to what may be causing it? 

Thanks, in advance,
Kevin Fonda

arendst, 2017-03-03:
You are missing at least file xdrv_snfled.ino

Check your sonoff folder against this repository...

kfonda, 2017-03-03:
The xdrv_snfled.ino file was there. It seems even stranger than I thought.  

- If I download the repository as a zip file and unzip it, it compiles fine.
- If I use `git clone https://github.com/arendst/Sonoff-Tasmota.git` or clone it with SourceTree, it fails to compile.
- If I use `git clone https://github.com/arendst/Sonoff-Tasmota.git` on a linux machine and copy the repository to my windows machine, it compiles fine.

The only difference I can find is that the code that compiles is in the Unix(LF) format and the code that fails is in the Windows(CRLF) format.
I'll try to install everything on a different windows 7 machine and see if it has anything to do with my configuration when I get a chance.

P.S. - I would like to thank you for providing this code. I'm amazed at just how much you've managed to cram into the tiny esp8266.

closed 2017-03-05 by arendst

ThomDietrich, 2017-03-06:
I've encountered same problem. git clone on a Windows machine will result in the above compilation error.

legotheboss, 2017-03-06:
Experiencing no problems on Windows 10, might be a compatibility issue on PlatformIO on older windows versions...

Chinchilla, 2017-03-06:
Not sure I should bother commenting on a closed issue, but this is a line ending conversion issue.

I've not looked into where the problem is exactly, but setting **core.autolf false**, in git will avoid doing the CRLF conversion and then the compile will succeed.

kfonda, 2017-03-07:
@Chinchilla Thanks for the tip, that solved the problem I was having.

ThomDietrich, 2017-03-07:
@arendst this should be solvable on your side, is it? I'd suggest to reopen the issue till then.

arendst, 2017-03-07:
@ThomDietrich I guess it's your local git install that needs a config change as documented here https://git-scm.com/book/be/v2/Customizing-Git-Git-Configuration

davidelang, 2017-03-07:
I'm not sure that .gitattributes is distributed as part of the repo, I think it's a local configuration.

ThomDietrich, 2017-03-07:
.gitattributes is part of the distributed repository, just as .gitignore.
@arendst I'm not sure if it will help. I have some experience with Windows/Linux line endings and git and by "solvable" I meant on repository side so users are not effected. (They shouldn't need to compile themselves but that's another story)
I'll do a few tests this evening and come back here.

ThomDietrich, 2017-03-07:
Very strange issue. The culprit was `sl_blank()` inside `xdrv_snfled.ino`.
git clone will on default windows system convert to crlf (something I normally disable). Converting the one file back solved the issue... so did the PR I just submitted. Having the block comment above the function is the correct way following most style guides anyhow...

The reason for this error riddles me. This is the minimal non-working example:

```cpp
void sl_blank(byte state)
/*
 *
 *
 */
{
```

Yes, it has to be at least a four line block comment 😄