arendst/Tasmota #148: Add a timer...

gemitro, 2017-03-04:
It would be nice to add a timer to the configuration menu.

arendst, 2017-03-05:
And what kind of timer do you want? Once a day at a certain time? Once after a certain time has elapsed?
What do want the timer to do? Switch one relay? Switch all relays? Just send a MQTT message?

Having answered these questions think about the code needed to implement and do not forget the user input validation...

Feel free to supply a pull request ;-)

gemitro, 2017-03-05:
Personally i use the sonoff basic as a light switch. So a timer that switch on & off the light at certain times every day would be nice. At the moment i use crontab & mosquitto to a raspberry pi3.

Thanks for the response.

arendst, 2017-03-05:
That's the correct way to go using an external manager like you do. I guess I leave it that way.

Greg2002, 2017-03-05:
Now we already have I2C with a number of sensors. We can connect to I2C RTC (DS3231) and have very good RTC, timer, alarms, real EEPROM for config and so on. And it will work when you loose or have not at all internet connection. Let it be!?

Greg2002, 2017-03-05:
It's realized in Russian project. Code: https://drive.google.com/open?id=0B7YhCK2ZNsYYcjg4VHhsTlNJODQ
and video (in Russian) https://www.youtube.com/watch?v=0gjqXPSK4yE

davidelang, 2017-03-05:
I was actually about to start to look for timer functionality to turn something 
on for X amount of time and then turn it off.

I'm happy to do this with an external manager (I'm using openhab), but haven't 
yet done much looking to see the easiest way to do this.

closed 2017-03-10 by gemitro

ChienTing, 2017-03-29:
If you can join the timing it is more perfect
Stand-alone device can be set when the time off
Do not add the host to do control

redviper2100, 2017-04-20:
Are there any updates regarding this timer? Thanks.

lobocobra, 2017-09-03:
ESPeasy has this in the code. You can program a little sketch there.
=> pro.... your device becomes more independent and even if your server is offline, it works
=> contra.... how are you go to keep this updated, with 40 devices all around?

hsmptg, 2017-09-16:
Hi
I think a timer in the sonoff could be helpfull.
I use a sonoff to control a valve of an irrigation system of my garden.
Since it is in the limit of the wifi range I am always afraid of it do not receive the turn off command after being received sucessfully the turn on command.
With a timer I could send a command like: turn on and after nn seconds turn it off.
This way even in the case of losing the wifi connection after turn on the valve I would have certain that the valve would turn off!
Gotcha?
Regards,
Hélio

davidelang, 2017-09-16:
already there, see the pulsetime command

flexiti, 2017-10-24:
I see a lot of resistance  ;) before implementing a timer like
[image]

Since we have time from internet, **the feature can work locally, on the device,** and can be programmed with MQTT
eq,
`AllTimer on/off  (default off) `
and eg.
`Timer1:{Time:12:00, OnOff:(on/off/nop), Relay: 1, Replay:(on/off)}`
Timer2:.....  etc. 
There could be eg. 8 timers
If `Replay:off`, this would be a one time action
If someone wants to splash you can add days of the week ;)

ps. As we have access to the internet we have set the time correctly ... and if there is no internet .... nothing can be controlled, Sonoff-tasmota is useless   ;)

By the way, Sonoff-Tasmota is a great, good job 👍  :)

davidelang, 2017-10-24:
network access != Internet access

On Tue, 24 Oct 2017, flexiti wrote:

> I see a lot of resistance  ;) before implementing a timer like
> [image]
>
> Since we have time from internet, **the feature can work locally, on the device,** and can be programmed with MQTT
> eq, [...]

flexiti, 2017-10-24:
I feel sorry for everyone who has a network without internet access 😉

davidelang, 2017-10-24:
most of the time it's deliberate. It narrows the attack range.

On Tue, 24 Oct 2017, flexiti wrote:

> I feel sorry for everyone who has a network without internet access 😉

flaotte, 2018-02-28:
I have timer request also. But it is a bit different timer I need. I want to have "maximum working time" timer. E.g. 4 hours. Just to have limit for devices that are not meant to be on 24/7.

**Requirements**:

- setup time in minutes;
- relay will be turned off after working for given time;
- if you will send on command during device is working, timer counter resets;
- drifting from exact timing is ok (+/-20%). It is not chronometer.

**optional**:
 - provide duration within on command. "switch on for 15 minutes";

I like to use default-on mode. It works fine with old light switches. You keep them on, and control light with sonoff, but you can always use hardware switch to control light or remove sonoff.
But... If something fails (power outage + server fails) when I on vacation etc I don't want my lights to be on for weeks.

Is it possible to get timer for maximum time on? You can use it for other purpose also.

flexiti, 2018-03-01:
#1091  meets your expectations

on-off-on, 2018-03-21:
Is there relly no way to implement in the "original" repository a simple timer function based on a electronic timer? :-// It would be a great feature for users like me to have "one" firmware to switch with commands from a controller e.g. FHEM and a simple timer function within the unit for areas without WLAN or LAN-access. I would like to have a timer for security reason during my holiday.... without connection to FHEM-controller... 
The forked timer doesn´t work and I´m unable to find the issue :-/

ascillato, 2018-06-22:
Sam9s

Please read the wiki. Timers are implemented. This is a very old and close  thread.

Thanks

sam9s, 2018-06-28:
ascillato, I dont need timer ... I know timer is there called pulstime, I need scheduling ... like the device should start on at certain time certain date and switch off certain time and date , with repeat schedule ..... I was able to find a fork of tasmota with schedule BUT looks like it works with MQTT. I dont want to run anything extra, just a scheduling feature with standalone tasmota on sonoff.

ascillato, 2018-06-28:
Hi

For timing in tasmota you have already implemented in the actual version:

- Pulsetime (time to turn off relays after they are turn on)
- Timing variables for rules:
1 - uptime
2 - time in seconds from midnight
3 - sunset
4 - sunrise
- Ruletimer (countdown timers for being used inside rules)
- And weekly Timers:
[image]

JeffJefferson, 2023-07-08:
It would be really great if you could add a duration parameter for automatically turn off the specified power output. Otherwise you always need a power on timer and a second timer for powering off.