arendst/Tasmota #152: port80

jkiape, 2017-03-04:
Change web port from 80 to ...any ???

davidelang, 2017-03-04:
in webserver.ino is the line:

       webServer = new ESP8266WebServer(80);

change the 80 to your desired port and I'll bet that it works.

David Lang

jkiape, 2017-03-04:
David thnx for your respond but unforchantly does not work.

davidelang, 2017-03-04:
well, if that doesn't work, you'll have to change the library

on my system it's:
/home/dlang/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h

change port=80 to whatever you want it to be.

but reading the header, the change to webserver.ino really should work.

David Lang

On Sat, 4 Mar 2017, jkiape wrote:

> Date: Sat, 04 Mar 2017 13:08:07 -0800
> From: jkiape <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fefc52511a8f42710078c506c0d2a43ef8497e48092cf0000000114d2eeb
>     792a169ce0ca16382@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: davidelang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Change port 80 (#152)
> 
> David thnx for your respond but unforchantly does not work.
>
>

jkiape, 2017-03-05:
Something strange happens. The server listens to the port (81) I gave (telnet 192.168.1.127 81) but it does not open by the web browser (http://192.168.1.127:81)

davidelang, 2017-03-05:
It's possible that your system has a firewall in place that prevents the browser 
from going to odd ports (it's a valid security concern)

why are you needing to run the webserver on a different port anyway? it's not 
like there is anything else on that IP address to compete with it.

David Lang

On Sun, 5 Mar 2017, jkiape wrote:

> Something strange happens. The server listens to the port (81) I gave (telnet 192.168.1.127 81) but it does not open by the web browser (http://192.168.1.127:81)

jkiape, 2017-03-05:
εxperimenting and ... playing.
Thank's any way,

davidelang, 2017-03-19:
see PR #232 

it turns out is was just the line in webserver.ino

closed 2017-03-19 by davidelang

godlich, 2018-04-08:
Honeslty, I am a beginner especially with tasmota. Do I have to change the file, recompile everything and upload the new code to my sonoff? Or can I do the changes online by connecting with tasmota via telnet?