arendst/Tasmota #156: Remote Access

jkiape, 2017-03-05:
Trying to gain remote access to sonoff I made a port forward to sonoff's local ip port 80.
When I am trying to connect from outside, the web server redirects to local ip. 
Any ideas ?

jkiape, 2017-03-05:
- [ ]

legotheboss, 2017-03-05:
I don't know if I would recommend this. If you want remote access, I would suggest using a public MQTT broker and conveying commands via that. Make sure you use strong credentials to prevent others from maliciously using your sonoff.

jkiape, 2017-03-06:
I'm aware of this process and I can do it relatively safe through Raspeverry pi (and port  forward) that I have.

Thank's

splodgemcroo, 2017-03-08:
I had the same problem, and tracked the issue down to the fact that the webserver component has a captive portal function (presumably to help with configuring the sonoff if it's in AP mode? not sure as I haven't used it in that way).

In order to disable redirecting to the local IP, i edited the webserver.ino file, found the captivePortal() function, and commented out the whole  if (!isIp(webServer->hostHeader())) block so that the function always returns false.

Hopefully I've explained that ok - let me know if not, and I'll try and elaborate.

jkiape, 2017-03-08:
It works! 
Thank you very much

arendst, 2017-03-08:
Could you make the below change on line 1358 in file webserver.ino and let me know if that solves your problem too?
From
```
  if (!isIp(webServer->hostHeader())) {

```
To
```
  if ((_httpflag == HTTP_MANAGER) && !isIp(webServer->hostHeader())) {
```

arendst, 2017-03-08:
Could you check if changing line 1358 in file webserver.ino from
```
  if (!isIp(webServer->hostHeader())) {
```
to
```
  if ((_httpflag == HTTP_MANAGER) && !isIp(webServer->hostHeader())) {

```
also solves your problem please?

jkiape, 2017-03-08:
>also solves your problem please?
Yes it does !

closed 2017-03-08 by arendst

gabryk91, 2018-04-21:
@arendst I'm also trying to connect to my tasmota from outside, I did all the necessary configuration on my router from an x port to port 80 of the tasmota, but I can't see the web server, I had searched in the webserver.ino for the suggested lines, but I can't find those (I guess they were changed in recent updates). Can you provide me some direction to solve this problem? I understand the warning about opening the tasmota to the outside world, but still I want to do it with all the precautions, thanks