arendst/Tasmota #158: Momentary switch with Toggle (Function Request)

TootToot168, 2017-03-06:
Not sure if this function is possible to be added into the current firmware or it is already there. 

It will helps to add physical control of the device without additional wiring. 

Setup a momentary switch (normally closed) before the "L" input of Sonoff
Whenever the momentary switch is triggered, Sonoff toggle the output state.

Basically the firmware has to detect whenever there is a change in the input power state, it will also change the output power state.

davidelang, 2017-03-06:
currently the options when power comes on are:

PowerOnState     |              | Show current relay power on state
PowerOnState     | 0 | off      | Keep relay(s) off after power on
PowerOnState     | 1 | on       | Turn relay(s) on after power on
PowerOnState     | 2 | toggle   | Toggle relay(s) on from last saved
PowerOnState     | 3            | (default) Turn relay(s) on as last saved

I would understand that 2 and 3 are opposites, so if you set it to 2, it will 
come on the opposite of what it was before.

If it doesn't, such a mode would be easy to add.

TootToot168, 2017-03-06:
where can these options be configured or determined ?

arendst, 2017-03-06:
The wiki might be a good starting point...

davidelang, 2017-03-06:
at the console in the web interface, via MQTT commands, or they can be set in 
user_config.h

TootToot168, 2017-03-07:
Hi i went through the wiki and understand in order to achieve this i need to set the PowerOnState to 2 so that it will toggle everytime the Sonoff is powered on. I tried to issue command in the console like "cmnd/sonoff/PowerOnState2" or "PowerOnState2" but it just return me result of stat/sonoff/RESULT = {"PowerOnState":3}. what is the exact command i need to issue in the console ?

davidelang, 2017-03-07:
poweronstate 2

k

TootToot168, 2017-03-07:
thank you very much

TootToot168, 2017-03-07:
poweronstate 2 at console works!! 
Able to toggle the light whenever there is change in power state of the sonoff but face some challenging using a normally closed Momentary switch.

When using a normally closed Momentary switch, the toggle of the light does not work well because the sonoff seems to "hold" power for 1 sec even after powered off. 
The open and close circuit via the Momentary switch happens too fast for the sonoff to acknowledge a change in the power state to toggle the light. When the normally closed Momentary switch is triggered, the light does turn off and comes back on immediately, the LED indicator on sonoff remains on.

Not sure if this is due to a hardware issue or there is a "delay" in the programming logic for the power state change check therefore the toggle of the light does not happen.

davidelang, 2017-03-07:
It's a hardware issue, the power supply on the sonoff takes some time to drop to 
too low a voltage for the system to operate when it looses power (the power 
supply taps from line voltage down to something in the 5-10v range depending on 
what your line voltage is, and then there's a switching power supply that drops 
it from there down to the 3.3v that the esp8266 wants, and it actually fails at 
an even lower voltage.

So if the relay is off, the LED is off, and the esp is consuming minimum power, 
it can take a noticable amount of time between the time you cut input power and 
when the system dies. If you are unlucky enough, you will just corrupt it's 
memory instead of it either surviving or dieing.

David Lang

closed 2017-03-08 by TootToot168