arendst/Tasmota #160: Sonoff restarts after received wrong MQTT message

ionciubotaru, 2017-03-06:
On version 4.0.0 sonoff restarts when receiving:

Topic: cmnd/sonoff_sub_birou 
Message: status

Message after restart:
tele/sonoff_sub_birou/INFO3
{"Started":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4021e700 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"} 

When proper MQTT format is used ( cmnd/sonoff_sub_birou/status ) everything works fine

arendst, 2017-03-06:
I cannot reproduce. If I enter
```
 mosquitto_pub -t cmnd/sonoff_sub_birou -m status
```
sonoff responds with
```
10:17:51 RSLT: DataCb Topic sonoff_sub_birou, Group 0, Index 1, Type <null>, Data status (STATUS)
10:17:51 MQTT: stat/sonoff_sub_birou/COMMANDS1 = {"Commands1":"Status, ...
```
as expected.

What information do you receive with logging option 3?

ionciubotaru, 2017-03-06:
I upload the firmware to a Sonoff Basic:
Firmware is compiled with:
`#define BE_MINIMAL     `
but I comment:
```
//#ifdef USE_DOMOTICZ
//#undef USE_DOMOTICZ                         // Disable Domoticz
//#endif
```
The command is:
`mosquitto_pub -h m21.cloudmqtt.com -p 16237 -u <username> -P <password> -t cmnd/sonoff_test -m status`

Serial output is:
```
00:00:00 APP: Project sonoff_test Sonoff (Topic sonoff_test, Fallback DVES_0CD6BD, GroupTopic sonoffs) Version 4.0.0
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Connecting to AP2 Public in mode 11N as sonoff_test-5821...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Checking connection...
00:00:04 Wifi: Connected
00:00:04 HTTP: Webserver active on sonoff_test-5821 with IP address 192.168.1.106
00:00:05 MQTT: Attempting connection...
00:00:06 MQTT: Connected
00:00:06 MQTT: tele/sonoff_test/LWT = Online (retained)
00:00:06 MQTT: cmnd/sonoff_test/POWER = 
00:00:06 MQTT: tele/sonoff_test/INFO1 = {"Module":"Sonoff Basic", "Version":"4.0.0", "FallbackTopic":"DVES_0CD6BD", "GroupTopic":"sonoffs"}
00:00:06 MQTT: tele/sonoff_test/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff_test-5821", "IPaddress":"192.168.1.106"}
00:00:06 MQTT: tele/sonoff_test/INFO3 = {"Started":"Software/System restart"}
00:00:06 RTC: (UTC) Mon Mar 06 15:08:54 2017
00:00:06 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:06 RTC: (STD) Sun Oct 29 03:00:00 2017
17:08:55 MQTT: stat/sonoff_test/RESULT = {"POWER":"OFF"}
17:08:55 MQTT: stat/sonoff_test/POWER = OFF
17:09:01 HTTP: Handle root
17:09:02 MQTT: tele/sonoff_test/STATE = {"Time":"2017-03-06T17:09:02", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":2, "SSID":"Public", "RSSI":68}}
17:09:04 HTTP: Handle config
17:09:06 HTTP: Handle Domoticz config
17:09:09 HTTP: Handle config
17:09:11 HTTP: Handle MQTT config
17:09:12 Wifi: Checking connection...
17:09:12 Wifi: Connected
17:09:13 HTTP: Handle config
17:09:17 HTTP: Handle Log config
17:09:32 Wifi: Checking connection...
17:09:32 Wifi: Connected
17:09:33 RSLT: Receive topic cmnd/sonoff_test, data size 6, data status

Exception (28):
epc1=0x4021e700 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1e70 end: 3fff2690 offset: 01a0

```

arendst, 2017-03-06:
Did it your way and still doesn't fail.

Please provide the output of command ``status 0``

ionciubotaru, 2017-03-06:
Output of `status 0` 
```
18:14:47 CMND: status 0
18:14:47 RSLT: Receive topic cmnd/sonoff_test/status, data size 1, data 0
18:14:47 RSLT: DataCb Topic sonoff_test, Group 0, Index 1, Type STATUS, Data 0 (0)
18:14:47 MQTT: stat/sonoff_test/STATUS = {"Status":{"Module":1, "FriendlyName":"Sonoff", "Topic":"sonoff_test", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":0, "SaveState":0, "ButtonRetain":0, "PowerRetain":0}}
18:14:47 MQTT: stat/sonoff_test/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff_test.ino.bin", "Uptime":0, "Sleep":0, "BootCount":2, "SaveCount":2}}
18:14:47 MQTT: stat/sonoff_test/STATUS2 = {"StatusFWR":{"Program":"4.0.0", "Boot":4, "SDK":"2.0.0(656edbf)"}}
18:14:47 MQTT: stat/sonoff_test/STATUS3 = {"StatusLOG":{"Seriallog":4, "Weblog":0, "Syslog":0, "LogHost":"stormbet.cometa.ro", "SSId1":"Clicknet-BB4A", "SSId2":"Public", "TelePeriod":300}}
18:14:47 MQTT: stat/sonoff_test/STATUS4 = {"StatusMEM":{"ProgramSize":438, "Free":500, "Heap":24, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":1024, "ProgramFlashSize":1024, "FlashChipMode":2}}
18:14:47 MQTT: stat/sonoff_test/STATUS5 = {"StatusNET":{"Host":"sonoff_test-5821", "IP":"192.168.1.107", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:0C:D6:BD", "Webserver":2, "WifiConfig":4}}
18:14:47 MQTT: stat/sonoff_test/STATUS6 = {"StatusMQT":{"Host":"m21.cloudmqtt.com", "Port":16237, "ClientMask":"DVES_%06X", "Client":"DVES_0CD6BD", "User":"ionut", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
18:14:47 MQTT: stat/sonoff_test/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 06 16:14:47 2017", "Local":"Mon Mar 06 18:14:47 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":2}}
18:14:47 MQTT: stat/sonoff_test/STATUS10 = {"StatusSNS":{"Time":"2017-03-06T18:14:47"}}
18:15:04 Wifi: Checking connection...
18:15:04 Wifi: Connected
```

here are my files [https://dl.dropboxusercontent.com/u/5288983/sonoff.zip](https://dl.dropboxusercontent.com/u/5288983/sonoff.zip) - all changes are located in sonoff\user_config_override.h

> BTW why are you running production in  BE_MINIMAL?

I'm not in production, I try to understand your great software

arendst, 2017-03-06:
And there you have it in StatusFWR: You are NOT running with core version Arduino/Esp version 2.3.0 as you are using SDK 2.0.0. I do currently not support another core version than 2.3.0 as there are a lot of dependencies I still have to work out. One of them is the much larger code base where I anticipate to use the BE_MINIMAL version to be able to swap OTA images.

So use the official core version 2.3.0 and all should be fine.

ionciubotaru, 2017-03-06:
Thank's a lot
Keep up the good work

closed 2017-03-06 by ionciubotaru