arendst/Tasmota #161: Production Mode

ionciubotaru, 2017-03-06:
Sonoff Touch is a great device, but has GPIO0 accessible to all users.
A child playing with the light switch can easily reset the Sonoff to it;'s default configuration loosing all saved variables like DomoticzIdx.

I suggest:
`#PRODUCTION 1 `

In production mode the following options will be disabled:
- serial log
- web server
- multiple clicks (over 3) - WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG
- reset to default on long push

davidelang, 2017-03-06:
The first two are already available.

the last two seem pretty easy to recover from (power cycle solves the first and 
the second can be handled in an automated way)

But I agree that an option to limit what can be done with the button isn't a bad 
idea.

On Mon, 6 Mar 2017, Ion wrote:

> `#PRODUCTION 1 `
>
> In production mode the following options will be disabled:
> - serial log
> - web server
> - multiple clicks (over 3) - WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG
> - reset to default on long push
>
>
>

arendst, 2017-03-08:
ButtonRestrict On/Off introduced to disable all  multi key presses above 2 and hold. See wiki commands.

closed 2017-03-08 by arendst