arendst/Tasmota #162: Reset - Boot loop

DerElch87, 2017-03-06:
Dear all,

I have several Sonoff Basic, but one piece makes a lot of problems.

Most of the time I had the following scenario:
Directly after flashing I can connect via WLAN but as soon as I change any settings which will force a restart I got stuck in a boot loop.

Meanwhile I can't even connect directly after flashing.

Below some logfiles.

Thanks for your help in advance

------------
00:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_964094, GroupTopic sonoffs) Version 3.1.16
00:00:00 Wifi: Connecting to AP1 Fritzi_die_Grosse in mode 11N as sonoff-0148...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
þ
------------
00:00:00 APP: Project sonoff Sonoff_2 (Topic sonoff, Fallback DVES_964094, GroupTopic sonoffs) Version 3.9.12
00:00:00 Wifi: Connecting to AP1 Fritzi_die_Grosse in mode 11N as sonoff-0148...

E
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
ø
------------
00:00:00 APP: Project sonoff test (Topic test, Fallback DVES_964094, GroupTopic sonoffs) Version 4.0.1
00:00:00 Wifi: Connecting to AP1 Fritzi_die_Grosse in mode 11N as test-0148...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
ü

arendst, 2017-03-07:
As it is only one out of your many sonoffs we can rule out my software right?
I would suggest you try to completely erase your sonoffs flash with a command like
```
esptool.py --port COM1 erase_flash
```
The program can be found at https://github.com/espressif/esptool

After that reflash my software and try again.

If it still fails I guess you have a broken sonoff.

davidelang, 2017-03-07:
On Tue, 7 Mar 2017, Theo Arends wrote:

> As it is only one out of your many sonoffs we can rule out my software right?
> I would suggest you try to completely erase your sonoffs flash with a command like
> ```
> esptool.py --port COM1 [...]

I wonder if he has some bad data in the config. If so, flashing it with a 
different value in CFG_HOLDER may help (one step shy of a complete erase)

David Lang

DerElch87, 2017-03-08:
Just a quick feedback from my side.
I tried both of  your ideas, means to erase the flash and also change the CFG_HOLDER value.

None of them worked.
The reset cause and boot mode number stays the same.

On top of it I also tried another mqtt software for the sonoff basic with the same result and failure message.

So yes, this unit is most probably defective.

Anyway, thanks for the quick support.

closed 2017-03-08 by DerElch87

tpaulic, 2017-10-17:
Dear All,

I have a similar problem. When I sent my MQTT to 'restart', he entered the boot loop and never came out.

I have just managed to solve the problem that I shut down the MQTT and now the device is working but I need MQTT to connect it with openHAB

Does anyone have an idea of ​​how to fix it?

tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff TH", "Version":"5.8.0", "FallbackTopic":"SonofTH16_01", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}

stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF

tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
tele/sonoff/INFO1 {"Module":"Sonoff TH", "Version":"5.8.0", "FallbackTopic":"SonofTH16_01", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/RESULT {"Restart":"Restarting"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}

Duesentr, 2018-01-16:
Hi there!

I have the same problem as described in the first post of DerElch87: one out of three sonoff basic gives me contineous boot loops.
Neither changing CFG_HOLDER nor a full erase of the flash with esptool.py solved the problem.

log from the serial port:

> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0769A9, GroupTopic sonoffs) Version 5.11.1
> 00:00:00 WIF: Connecting to AP1 SeinBruder-IOT in mode 11N as sonoff-2473...
> 08:00:11 WIF: Connected
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> 
> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0769A9, GroupTopic sonoffs) Version 5.11.1
> 00:00:00 WIF: Connecting to AP1 SeinBruder-IOT in mode 11N as sonoff-2473...
> 00:00:08 WIF: Connected
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> 
> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0769A9, GroupTopic sonoffs) Version 5.11.1
> 00:00:00 WIF: Connecting to AP1 SeinBruder-IOT in mode 11N as sonoff-2473...
> 00:00:04 WIF: Connected
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> 
> 

apparently DOA...

lobradov, 2018-01-16:
Did you, by chance, set the "cmnd/sonoff/restart" as retained message? :)

webtop, 2018-01-17:
The reset is the intended behavior by the watchdog timer. The issue is caused by a memory error, or a communication problem. The watchdog is doing what is supposed to do, which is to reset the device when it’s not responding.

lobradov, 2018-01-17:
@webtop - looking only at @tpaulic case: 
Isn't watchdog supposed to be `rst cause:4` and not `1`, which is "normal boot"?
(that's also a puzzle in this case - if it was reset by retained message, it would have been `3`).

Anyway, for reference:
```
reset causes:
    0: 
    1: normal boot
    2: reset pin
    3: software reset
    4: watchdog reset

boot device:
    0:
    1: ram
    3: flash
```

For OP (@DerElch87) I agree - it's definitely watchdog.

Tech4Ever, 2018-01-20:
Same Problem here!
The Pow worked once, I was able to create an AP and connect to it. Unfortunately it didn´t connect to my network, so I was looking into the serial output and there I found this log:

`00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A02DAF, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP2 _MYWLAN_ in mode 11N as Pow...
00:00:03 WIF: Connected

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld`

roguestreak, 2018-01-21:
I've got an identical to Tech4Ever, new unflashed POW, I've been tearing my hair out for half the day trying to figure this out as my previous flash of an S22 worked OK (5.10), but I also updated the ESP library in arduino this morning and things got pear-shaped so I completely deleted the arduino software and started from scratch.

> 00:00:00 Project sonoffpow Sonoff (Topic sonoffpow, Fallback DVES_4EBCE9, GroupTopic sonoffs) Version 5.11.1
> 00:00:00 WIF: Connecting to AP1 MYWIFIPLACE in mode 11N as sonoffpow-7401...
> 00:00:05 WIF: Connected
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
Rinse and repeat

roguestreak, 2018-01-21:
Addendum; just flashed one of the basics and and it appears to be working (though mqtt isn't connecting at the moment...)

Tech4Ever, 2018-01-21:
I found the reason of my problem:

in "user.config" I missed to change the IP from "192.168.2.1" (e.g. gateway) to "192.168.1.1". I missed the "2" thought it´s a "1"...

roguestreak, 2018-01-21:
I just tried a complete flash erase with esptool and a reflash, and I'm still getting the identical rebooting error of the watchdog timer?
David I don't believe it would be an mqtt thing, as it turns out I had my authentication details wrong in the config; when I flashed the Basic it wouldn't connect and I eventually figured out the username was wrong.  The Basic is now working fine with the identical firmware and settings uploaded.

arendst, 2018-01-24:
I see the reboot watchdog appear but only if I compile with:

- Esp8266/Arduino v2.4.0
- Arduino IDE
- Arduino IDE - Tool - LwIP Variant: V2 Prebuilt (MSS=536)

So I've updated the wiki - Arduino IDE part to select LwIP Variant: 1.4 Prebuilt which works just fine.

roguestreak, 2018-01-28:
Thanks Theo, I just recompiled and uploaded with that setting and so far it seems to be working as intended.

bellaf, 2018-02-14:
Theo, many thanks, this LwIP Variant change has fixed the same issue on my recenyl aquired Node-MCU 12E... superb!  Keep up the great work...

Duesentr, 2018-03-11:
Thank you, Theo!
This worked for me, too. I just revived the sonoff basic that I assumed to be DOA. Strange that it worked for three out of four sonoff basic with the"wrong" LwIP Variant setting, But I am happy to have an additional device now.

lobocobra, 2018-09-17:
Add on... I could revive my sonoff, but had to do the following...
1) erase with above tool
2) connect to serial and make a "reset 2)
3) ssid xxx and password xxx to serial (as I could not connect to wifi
4) it will restart, de-connect vcc and re connect
5) in serial enter "upgrade 1), leave it 5 min alone
6) now it went back to life and webserver worked again, rest is as usual